vlachoudis/bCNC #93: Laser cutter option

dougmany, 2015-08-22:
Do you have any interest in a laser cutter option in the config file?  I bought a Raspberry Pi CNC to replace the crappy control board on a Chinese laser cutter.  I got bCNC to work by replacing z changes in the code with m3 and m5 commands like this:

-			block.append("g0 %s"%(self.fmt("z",self.safe)))
+			block.append("M5")

-			block.append("g1 %s %s"%(self.fmt("z",z), self.fmt("f",self.feedz)))
+			block.append("M3")

vlachoudis, 2015-08-22:
Very nice idea. I don't have a laser cutter but I will add an option in the config for lasercutter and modify the code accordingly.

dougmany, 2015-08-23:
I am looking forward to it.  Great job on this project.  It has really helped me out.

vlachoudis, 2015-09-13:
In the new interface (just uploaded) I have an option in the settings for the lasercutter.

closed 2015-09-13 by vlachoudis