vlachoudis/bCNC #95: Using it with no connection

onekk, 2015-08-26:
There is a simple manner to set BCNC to output the machine command to a file, making possible to use it without a machine, I have my machine in a distant place and for experimenting, i need to test on a computer not connected to grbl.

Thanks

Carlo D.
Onekk

vlachoudis, 2015-08-26:
If I understand correctly you are asking on how to output bCNC output to a file.
You would need to change the "def open(" routine in the bCNC.py and instead of creating a serial object you can change it to self.serial = open(device,"w")
I didn't try but except some serial handling all other things should work.

I wanted to do the same, for testing not to have to go to the garage, and I've the grblsim without any success.

onekk, 2015-08-26:
I will try this evening or night, or maybe if I'm too tired tomorrow, thanks.

Carlo D.

onekk, 2015-08-27:
I didn't figure out how to modify the code, 
You say to modify the "def open(" routine in the bCNC.py and instead of creating a serial object you can change it to self.serial = open(device,"w")

' 	def open(self, device, baudrate):
		if MOCK_SERIAL > 0:
                        self.serial = open(device,"w")

			# do something
		else:
			try:
				self.serial = serial.Serial(	device,
								baudrate,
								bytesize=serial.EIGHTBITS,
								parity=serial.PARITY_NONE,
								stopbits=serial.STOPBITS_ONE,
								timeout=0.1,
								xonxoff=False,
								rtscts=False)
				# Toggle DTR to reset Arduino
				self.serial.setDTR(0)
				time.sleep(1)
				CNC.vars["state"] = "Connected"
				CNC.vars["color"] = STATECOLOR[CNC.vars["state"]]
				self.state.config(text=CNC.vars["state"],
						background=CNC.vars["color"])
				# toss any data already received, see
				# http://pyserial.sourceforge.net/pyserial_api.html#serial.Serial.flushInput
				self.serial.flushInput()
				self.serial.setDTR(1)
				self.serial.write("\r\n\r\n")
				self._gcount = 0
				self._alarm  = True
				self.thread  = threading.Thread(target=self.serialIO)
				self.thread.start()
				return True
			except:
				self.serial = None
				self.thread = None
				tkMessageBox.showerror("Error opening serial",
						sys.exc_info()[1],
						parent=self)
			return False`

onekk, 2015-08-27:
Sorry. no luck in following your suggestion, and no luck in running grbl-sim (permission problem with the fake port) so no luck at all,

BUT I've found a fake pyserial module and trying to get it working, what are te protocol that bCNC is following to communicate with the arduino? or better see my github page https://github.com/onekk/bCNC the file fakeSerial.py is supposed to feed the serial with the data in the self._data variable, but i can't get bCNC stop invoking readline() and the buffer is emptyed, i suppose we can tweak the readline() function to act as a fake GRBL supplying the right string to bCNC and svae the output to a file, but i need to get some info in how bCNC work communicating with GRBL.

vlachoudis, 2015-08-29:
Thanks, its good to know.

Harvie, 2018-10-29:
It's not only about faking serial connection, we need to fake grbl status reports.

zio-pietro, 2019-11-20:
Good to know. I've the same "problem". Develop box here (at home) w/o CNC mill, workstation with another PC and CNC mill 5 km away. I'll take a look at onekk's solution, it promises well. I must also try grbl-sim, maybe it will be the right way. I will inform you soon.

Regards

Harvie, 2019-11-21:
Look into the "tests" directory in repository. There are several ways to do this.
Or just buy chinese arduino for $1 and flash real grbl into it. You don't need whole machine, just the arduino.

zio-pietro, 2019-11-22:
@harvie: yeah, I did so. Cheap chinese card, flashed grbl, card's USB chip (USB/serial converter) only lasted another 10 minutes and then passed away to the silicon heaven. The card isn't even usable as a paperweight, too light, 35 grams only :(  Not in vain one said: "I'm not so rich that I can afford to buy cheap things"