vlachoudis/bCNC #101: Supported Python version(s)?

justinclift, 2015-09-09:
Looking at the README, it doesn't mention which version(s) of Python are supported.  eg 2.x, 3.x, etc.

Asking because I'm stuck with using Windows for my CNC controlling PC (for now), so trying to figure out which Python download to grab. :wink:

Effer, 2015-09-09:
Hi, I suggest use python 2.7
Have a look at https://github.com/vlachoudis/bCNC/wiki/OS

justinclift, 2015-09-09:
Thanks. :smile:  It'd be a good idea to have this in the README, so I'll probably submit a PR for it later today/tomorrow after testing. :)

vlachoudis, 2015-09-13:
For the moment only the v2 is operational. I am slowly adapting the code to be used with v3, but there are some problems like unicode doesn't exist as type in python v3, since all strings now are unicode, however the serial accepts only bytes. I hope I will have a common solution for both versions quite soon.

closed 2015-09-13 by vlachoudis

paulvdhoeven, 2019-04-23:
I am curious about the state of using Python 3 for bCNC.
Python 2 is getting pretty obsolete, and support for it will end by the end of this year (2019).

Harvie, 2019-04-23:
We are all well aware of this. But we've found it's not exactly easy to port the code to python 3.
It's more than possible, but it will take some time and experience. I've never really done this. I've tried to port the bCNC to python 3 several times, but i've always failed on something related to unicode strings and tkinter.

I want to finish work on 0.9.15 release and then it will be possible to freeze bCNC development and port it to P3, but i think we need someone who is familiar with tkinter and unicode on P3.