vlachoudis/bCNC #103: Spindle on/off button not working, sort of

Ozzy62, 2015-09-13:
When opening bCNC the spindle on/off button does not work. M3/M5 on the commandline works ok.
Opening a file and starting and stopping it again makes the spindle on/off button working. Is that a bug ?.

vlachoudis, 2015-09-13:
I've just updated the interface with a completely new look and functionality. And this problem should be not be there. Could you please confirm?

Ozzy62, 2015-09-13:
Tried the latest version but it will not run , complaining about:
   File "./bCNC.py", line 8
    <<<<<<< HEAD
     ^
SyntaxError: invalid syntax

vlachoudis, 2015-09-13:
Please re-download again. During the merge of the two branches I've got this conflicts inside the files, but it was corrected some minutes ago.

Ozzy62, 2015-09-13:
Sorry to say that the problem is still there. I am running bCNC on a Raspberry Pi 2, if that makes a difference.

vlachoudis, 2015-09-13:
Just for confirmation, clicking the Spindle button sends a M3/M5 command it will not start the spindle if the RPM on the slider is not set.

Ozzy62, 2015-09-13:
I do not see any M3/5 commands in the terminal. But loading a file and starting/stopping will make M3/5 commands from the spindle button visible in the terminal window. And yes I do have set a speed on the spindle.

vlachoudis, 2015-09-13:
Ok I will check on my machine

vlachoudis, 2015-09-13:
I've just tested it on my machine, and I it works Ok. Does your grbl goes on Hold state when it reboots and then bCNC doesn't send any command. Did you unlock first?

Ozzy62, 2015-09-13:
Sorry. I did not do the unlock thingy. It works perfectly ok when having unlocked. 
Have you any clue on why I cant get the latest, and greatest, version to work ?.

vlachoudis, 2015-09-13:
Ok, no problem, good to know what was the issue :)

Can you be more explicit what is the problem with the new version?

Ozzy62, 2015-09-13:
When executing sh bCNC from a terminal i get the following:
File "./bCNC.py", line 8
<<<<<<< HEAD
^
SyntaxError: invalid syntax

Doing the same on a older version works ok, the program starts ok.

vlachoudis, 2015-09-13:
As I mentioned before, please re-download the program either the zip from github or with the git clone https://github.com/vlachoudis/bCNC

closed 2015-09-14 by vlachoudis