vlachoudis/bCNC #107: M6 / Change tool

mandrav, 2015-09-18:
Hi,

I really like the new look and features of the app!
Specifically, the new "change tool" functionality is exactly what I was waiting for but it looks like it still has some rough corners :) (or it may be my ignorance).

I 've noticed the following two issues with this:

1. Change tool without running a program
    - W/out running a job, I start the "change tool" procedure.
    - the tool moves to changeXYZ and enters feed-hold
    - I change the tool and press "pause" to resume (no physical keys currently on my cnc) (btw, is there a shortcut for the "pause/resume" key?)
    - the tool moves to probeXYZ and then probes down
    - the tool moves to changeXYZ, stays there and the UI is greyed-out (except for the "pause" and "stop" buttons
    - I don't know what it's waiting for at this point nor I can do anything to make it continue other than pressing the "stop" button so the UI is re-enabled...

2. Change tool while running a program
    - I 've set the policy to "ignore M6 commands"
    - when an M6 is encountered in the program, "error unsupported command" is displayed and it enters feed-hold...
    - the same happens if I manually enter "M6" in the command editbox
    - looks like the "ignore M6 commands" policy is not implemented?

Feel free to point out what I 'm doing wrong. This may very well be my mistake from not knowing how to work with this new functionality :)

vlachoudis, 2015-09-18:
I believe both are bugs that needs to be addressed.
1. it seems that the line counting is incorrect. I have to check
    Concerning the pause you can either assign a function key to it, Go to "Tools -> Shortcuts and assign any short cut the command "PAUSE", I've just added also the feedhold and resume (I will push it in a while)
2. I will check it

mandrav, 2015-09-18:
Great :)

Btw, regarding the second point, I forgot to mention that the "manual tool change" policy is also not respected...

vlachoudis, 2015-09-18:
@mandrav
1. apparently there was a %wait to idle command inside the code and the setting of the workspace was blocking. I don't understand why but removing it, fixed it. Download the new master and should work.
I forgot to mention that you can add also the Run/Pause/Stop buttons in the Probe section in your in
2. I made a small test and it works for me. Can you explain me a bit more in detail what do you see.

mandrav, 2015-09-18:
1.
I tried "change tool" and it worked.
I tried "calibrate" and it did it again (stopped after calibration and I had to "stop" to re-enable UI).
I tried "calibrate" again and it worked.

2.
Weird... it now doesn't say "error unsupported command" but instead it enters feed-hold. Still, after pressing "resume", it doesn't move to the tool change position nor it tries to probe the "new" tool...

mandrav, 2015-09-19:
I think I found the issue with the tool change functionality.
Looking at the code (toolChange function), it looks to me like it expects to find the Tn command maybe at the same line with M6?
My gcode generator puts the Tn a couple of lines before the M6 command so that's why it failed. I did a crude hack to remember the last seen Tn command and pass that along with 'cmds' in toolChange() and that seems to did the trick :)
One thing I noted though was that after the tool change, my Z feedrates remained low from the probing (25mm/m).

Anyway, here's the file I 've been playing with:

G21
G90
G94
F75.00
G00 Z1.0000
M03
G04 P1
G00 Z1.0000
T1
M5
M6
(MSG, Change to tool dia=1.0160)
M0
M03
G00 X29.2100Y9.1440
G01 Z-1.7000
G01 Z1.0000
G00 X29.2100Y6.6040
G01 Z-1.7000
G01 Z1.0000
G00 X29.2100Y4.0640
G01 Z-1.7000
G01 Z1.0000
G00 Z1.0000
T2
M5
M6
(MSG, Change to tool dia=1.3210)
M0
M03
G00 X6.6040Y9.1440
G01 Z-1.7000
G01 Z1.0000
G00 X6.6040Y4.1440
G01 Z-1.7000
G01 Z1.0000
G00 X0.0000Y0.0000
M05

vlachoudis, 2015-09-19:
@mandrav indeed I was expecting to see the T and M6 in the same line, thanks for spotting it.

vlachoudis, 2015-09-19:
@mandrav concerning the first problem. I could not see the reason when I wait that the buffer is destroyed. I am fighting with it right now :)

mandrav, 2015-09-19:
@vlachoudis about the M6 and Tn in different lines, will you fix it at some point or should I post a pull request with my hack (which you could improve ;))?

Thank you for the hard work :)

vlachoudis, 2015-09-19:
@mandrav I've pushed the change that remembers the tool number from a previous line as well the last feed rate to restore it after the tool change.

mandrav, 2015-09-19:
Now it's working great, thanks!

A couple of notes for you to decide if they need fixing:

After probing, the tool goes up to changeZ, first starts spinning and then enters feed-hold. Not a problem, it's just weird. I would expect it to start spinning after feed-hold or not entering feed-hold at all.

As an enhancement, it 'd be nice if it would return to the last position before the tool-change in rapid mode. Because I 'm testing it with a drill-file, the drill feed is slow (75mm/m). So, while it goes back to XY fast (rapid mode), the descend to the workpiece is painfully slow.
Would it be possible to also go to last Z (before tool change) with a rapid move?

As I said above, the feature now works for me as advertised. These two notes I made above are purely for enhancement. If you think it's better, you can close this issue and I can create a new one for the enhancements...

vlachoudis, 2015-09-19:
The tool change macro starts the spindle at the end before resuming the gcode. The feed-hold is coming from your gcode m0 command after the m6.

Running your g-code I came to the same conclusion with you. The program has to remember and restore the gantry position with a rapid movements at the end of the tool change cycle.

mandrav, 2015-09-19:
Yes, that M0, didn't notice it...

Thanks again.

vlachoudis, 2015-09-20:
@mandrav I've just added the restoring of the X/Y and then Z position with rapid motion
so it can save a bit of time

mandrav, 2015-09-20:
Thanks @vlachoudis , I 'll be sure to check it out :)

mandrav, 2015-09-21:
I confirm everything works as expected.
Issue closed :+1:

closed 2015-09-21 by mandrav