vlachoudis/bCNC #109: Statistics broken if reversing gcode order

mandrav, 2015-09-23:
In the editor tab, there's a button named "Reverse" which states that it "reverses direction of selected gcode blocks".
If you click this button, the statistics are messed up.
For example, I have a simple gcode file with one main block (a pocket).
Upon opening it, statistics say it needs ~11 minutes.
When I reverse the main block, statistics say it needs 0.66 minutes :)

I can post a sample file if needed, but it is very easily reproducible. If you do want a sample though, just ask :)

vlachoudis, 2015-09-23:
The reverse button is reversing the gcode order, but only for the first z-pass. If you have multiple z-steps in this pocket it will collapse them to a single pass, with a total path/time much sorter. Then you have to apply a cut operation again (from the tools).
I've a tried on a single z-path and the reverse doesn't change the stats. Can you confirm that this is the problem?

mandrav, 2015-09-23:
Single z pass exhibiting the problem here:

G21
G90
G94
F800.00
G00 Z1.0000
M03
G04 P1
G00 X1.5000Y1.5000
G01 Z-0.5000
G01 X1.5000Y98.5000
G01 X137.5000Y98.5000
G01 X137.5000Y1.5000
G01 X1.5000Y1.5000
G01 X3.3000Y3.3000
G01 X3.3000Y96.7000
G01 X135.7000Y96.7000
G01 X135.7000Y3.3000
G01 X3.3000Y3.3000
G01 X5.1000Y5.1000
G01 X5.1000Y94.9000
G01 X133.9000Y94.9000
G01 X133.9000Y5.1000
G01 X5.1000Y5.1000
G01 X6.9000Y6.9000
G01 X6.9000Y93.1000
G01 X132.1000Y93.1000
G01 X132.1000Y6.9000
G01 X6.9000Y6.9000
G01 X8.7000Y8.7000
G01 X8.7000Y91.3000
G01 X130.3000Y91.3000
G01 X130.3000Y8.7000
G01 X8.7000Y8.7000
G01 X10.5000Y10.5000
G01 X10.5000Y89.5000
G01 X128.5000Y89.5000
G01 X128.5000Y10.5000
G01 X10.5000Y10.5000
G01 X12.3000Y12.3000
G01 X12.3000Y87.7000
G01 X126.7000Y87.7000
G01 X126.7000Y12.3000
G01 X12.3000Y12.3000
G01 X14.1000Y14.1000
G01 X14.1000Y85.9000
G01 X124.9000Y85.9000
G01 X124.9000Y14.1000
G01 X14.1000Y14.1000
G01 X15.9000Y15.9000
G01 X15.9000Y84.1000
G01 X123.1000Y84.1000
G01 X123.1000Y15.9000
G01 X15.9000Y15.9000
G01 X17.7000Y17.7000
G01 X17.7000Y82.3000
G01 X121.3000Y82.3000
G01 X121.3000Y17.7000
G01 X17.7000Y17.7000
G01 X19.5000Y19.5000
G01 X19.5000Y80.5000
G01 X119.5000Y80.5000
G01 X119.5000Y19.5000
G01 X19.5000Y19.5000
G01 X21.3000Y21.3000
G01 X21.3000Y78.7000
G01 X117.7000Y78.7000
G01 X117.7000Y21.3000
G01 X21.3000Y21.3000
G01 X23.1000Y23.1000
G01 X23.1000Y76.9000
G01 X115.9000Y76.9000
G01 X115.9000Y23.1000
G01 X23.1000Y23.1000
G01 X24.9000Y24.9000
G01 X24.9000Y75.1000
G01 X114.1000Y75.1000
G01 X114.1000Y24.9000
G01 X24.9000Y24.9000
G01 X26.7000Y26.7000
G01 X26.7000Y73.3000
G01 X112.3000Y73.3000
G01 X112.3000Y26.7000
G01 X26.7000Y26.7000
G01 X28.5000Y28.5000
G01 X28.5000Y71.5000
G01 X110.5000Y71.5000
G01 X110.5000Y28.5000
G01 X28.5000Y28.5000
G01 X30.3000Y30.3000
G01 X30.3000Y69.7000
G01 X108.7000Y69.7000
G01 X108.7000Y30.3000
G01 X30.3000Y30.3000
G01 X32.1000Y32.1000
G01 X32.1000Y67.9000
G01 X106.9000Y67.9000
G01 X106.9000Y32.1000
G01 X32.1000Y32.1000
G01 X33.9000Y33.9000
G01 X33.9000Y66.1000
G01 X105.1000Y66.1000
G01 X105.1000Y33.9000
G01 X33.9000Y33.9000
G01 X35.7000Y35.7000
G01 X35.7000Y64.3000
G01 X103.3000Y64.3000
G01 X103.3000Y35.7000
G01 X35.7000Y35.7000
G01 X37.5000Y37.5000
G01 X37.5000Y62.5000
G01 X101.5000Y62.5000
G01 X101.5000Y37.5000
G01 X37.5000Y37.5000
G01 X39.3000Y39.3000
G01 X39.3000Y60.7000
G01 X99.7000Y60.7000
G01 X99.7000Y39.3000
G01 X39.3000Y39.3000
G01 X41.1000Y41.1000
G01 X41.1000Y58.9000
G01 X97.9000Y58.9000
G01 X97.9000Y41.1000
G01 X41.1000Y41.1000
G01 X42.9000Y42.9000
G01 X42.9000Y57.1000
G01 X96.1000Y57.1000
G01 X96.1000Y42.9000
G01 X42.9000Y42.9000
G01 X44.7000Y44.7000
G01 X44.7000Y55.3000
G01 X94.3000Y55.3000
G01 X94.3000Y44.7000
G01 X44.7000Y44.7000
G01 X46.5000Y46.5000
G01 X46.5000Y53.5000
G01 X92.5000Y53.5000
G01 X92.5000Y46.5000
G01 X46.5000Y46.5000
G01 X48.3000Y48.3000
G01 X48.3000Y51.7000
G01 X90.7000Y51.7000
G01 X90.7000Y48.3000
G01 X48.3000Y48.3000
G00 Z1.0000
G00 Z1.0000
G00 X0Y0
M05

vlachoudis, 2015-09-23:
That was fast :). Indeed there is a problem which this one.

vlachoudis, 2015-09-23:
I see. the reverse operation stripped the G4 P1 and it added the default feed of the current material.

mandrav, 2015-09-23:
No, *this* was fast :)

closed 2015-09-24 by vlachoudis