vlachoudis/bCNC #110: G83 macro espansion request

antoniosap, 2015-09-25:
Hi there, 
the G83 is a common g-gode for Peck Drilling Cycle, 
GRBL not have this code for lack of space.

Now, i run a post processor in perl for expansions in a cicle of G0/G1

How i change python code for this enhancement ?

vlachoudis, 2015-09-25:
Hi, you have to modify the "def compile" method of CNC.py
search in the method for the toolChange() handling and it has to
be done something similar. If you can convert your perl
to python, you could send me and I can add it

antoniosap, 2015-09-26:
ok. i am working on this.

antoniosap, 2015-09-27:
Hi, some support, please. Where is the current XYZ ?
CNC.vars["wx"],CNC.vars["wy"],CNC.vars["wz"] ??

vlachoudis, 2015-09-27:
@antoniosap correct the wx,wy,tz holds the last reported position with ?
if you want to know the current location accurately you need to halt the movement, %wait that the machine is idle and then use the wx,wy,wz position

antoniosap, 2015-09-28:
i need only the static position from syntax. not real time position. 
example:
N110 G90 Z0
N120 G0 G90 G54 X60.942 Y59.291
N130 S10000 M3
N140 G43.1 Z33.
N150 G98 G83 Z-1.202 R18. Q10. F50. (open macro)
N160 Y84.691
N170 Y110.091
N180 Y135.491
...
G80 (close macro)

The G83 macro code (on testing) is injected as..(from my debug)
macroG8X entry
['N150', 'G98', 'G83', 'Z-1.202', 'R18.', 'Q10.', 'F50.']
--->
['g0 x0 y0 z0', 'g0 z18', 'g1 z8 f50', 'g0 z18', 'g0 z8.01', 'g1 z-1.202 f50', 'g0 z18']
macroG8X exit

resulting: g0 x0 y0 z0 (wrong initial coordinates), G0 X60.942 Y59.291 Z33 (right initial coordinates)

i expect this espansion:
['g0 x60.942 y59.291 z33', 'g0 z18', 'g1 z8 f50', 'g0 z18', 'g0 z8.01', 'g1 z-1.202 f50', 'g0 z18']

where is the last static position of X,Y,Z ? 

the correct code expanded, for reference, is:
N110 G90 Z0  (N110 G90  Z0)
N120 G0 G90 G54 X60.942 Y59.291  (N120 G0 G90 G54 X60.942 Y59.291)
N130 S10000 M3  (N130 S10000 M3)
N140 G43.1 Z33.  (N140 G43.1 Z33.)
N150 G0 X60.942 Y59.291 Z33 
N151 G0 Z18 
N152 G1 Z8 F50 
N153 G0 Z18 
N154 G0 Z8.01 
N155 G1 Z-1.202 F50 
N156 G0 Z18  (N150 G98 G83 Z-1.202 R18. Q10. F50.)
N166 G0 X60.942 Y84.691 Z33 
N167 G0 Z18 
N168 G1 Z8 F50 
N169 G0 Z18 
N170 G0 Z8.01 
N171 G1 Z-1.202 F50 
N172 G0 Z18  (N160 Y84.691)
N182 G0 X60.942 Y110.091 Z33 
N183 G0 Z18 
N184 G1 Z8 F50 
N185 G0 Z18 
N186 G0 Z8.01 
N187 G1 Z-1.202 F50 
N188 G0 Z18  (N170 Y110.091)
N198 G0 X60.942 Y135.491 Z33 
N199 G0 Z18 
N200 G1 Z8 F50 
N201 G0 Z18 
N202 G0 Z8.01 
N203 G1 Z-1.202 F50

vlachoudis, 2015-09-28:
I've misunderstood you need the last x,y,z coordinates that was parsed. Normally the code is not parsing them before submitting (just to save time) unless if autolevel is requested. I can modify the code to parse them (or you can do). You would to call  the self.cnc.processPath(cmds) and the cnc.motionPathEnd() also in the the case with no autoleveling (without the motonPath). probably an else: would do the job. Then the x,y,z you can have from the self.cnc.x,   self.cnc.y...

If you want I can do it, but from tomorrow afternoon

antoniosap, 2015-09-29:
Hello, x,y,y stay at 0,0,0. attachjed is the code pre-alpha with debug code. i apply you suggestions. dropbox file at: https://www.dropbox.com/s/oser2x6kh71pdbs/CNC.py?dl=0

vlachoudis, 2015-09-29:
I will upload in later in the evening a new version that updates correctly the x,y,z variables inside the compile function

vlachoudis, 2015-09-30:
@antoniosap because I modified the CNC.py for the new toolChange() routine I cannot upload it yet. However if you make the following change it should work.
Inside the "def compile" of the GCode class the last method in the file
locate the "if autolevel:" and append an else branch as below
				else:
					self.cnc.processPath(cmds)
					self.cnc.motionPathEnd()

antoniosap, 2015-10-01:
changing as this, display 0,0,0 always. 

				if autolevel:
....
				else:
					self.cnc.processPath(cmds)
					xyz = self.cnc.motionPath()
					self.cnc.motionPathEnd()
					print self.cnc.x
					print self.cnc.y
					print self.cnc.z

vlachoudis, 2015-10-01:
You can checkout the latest CNC.py that works for me. Also what you mentioned should work, if not can you send me your test file, to understand what is the problem.

antoniosap, 2015-10-02:
after last checkout xyz = 0,0,0 this is the dropobox files:
*.NC the G83 macro
*.out.nc the G83 expanded by perl
CNC.py the prg and debug messages

https://www.dropbox.com/sh/ndip27hxe96u0qu/AAANcAemd17Qw1tSm_pDpf9Aa?dl=0

vlachoudis, 2015-10-03:
thanks @antoniosap. I see what is the problem. In the method processPath() and pathEnd() do not process at all the g83, g98. I will do later today, first I have to understand how the g83 works :)

vlachoudis, 2015-10-03:
Ok I've uploaded a new version that should take care of updating the x,y,z and as well return the r,q,l... values in the self.rval, self.qval....
However I am not sure if it is correct.

vlachoudis, 2015-10-04:
@antoniosap last night I read about the drill cycles and I've corrected the motion variables inside the CNC.py. So now it can display the canned cycles.
The variables you would need are the following
self.cnc.x, self.cnc.y, self.cnc.z  contain the last x,y,z position before entering the canned cycle
self.cnc.dx, dy, dz the increment
self.cnc.gcode (the current gcode)
self.cnc.lval, self.cnc.rval, self.cnc.qval
Please check the motionPath and the motionEnd for the correct use of the variables
The call to your routine should be done in the compile(), just before the motionEnd() call. Also for consistency it would be nice for check if the motionEnd returns correctly the x,y,z as your routine will return after the expansion.

I've tested the plotting with all the examples in linuxcnc and it works ok.

antoniosap, 2015-10-04:
wow. today i look this update

vlachoudis, 2015-10-04:
I just pushed one more update to work with your file

antoniosap, 2015-10-05:
i am testing. i added some debug print on '# Canned cycles should come here', however the coordinate processing is a least delayed of 1 instruction line. the display is OK.

the complete code and sample is here: https://www.dropbox.com/sh/m21o1wmrnhswcil/AABu5gECP0tHj162oN724rbwa?dl=0

test log.

CMDS1: ['N110', 'G90', 'Z0']
gcode1 None
0
0
0
CMDS1: ['N120', 'G0', 'G90', 'G54', 'X60.942', 'Y59.291']
gcode1 54
0
0
0
CMDS1: ['N130', 'S10000', 'M3']
gcode1 None
0
0
0
CMDS1: ['N140', 'G43.1', 'Z33.']
gcode1 43
0
0
0
CMDS1: ['N150', 'G98', 'G83', 'Z-1.202', 'R18.', 'Q10.', 'F50.']
gcode1 83
0
0
0
CMDS2: ['N150', 'G98', 'G83', 'Z-1.202', 'R18.', 'Q10.', 'F50.']
gcode 83
0
0
0
CMDS1: ['N160', 'Y84.691']
gcode1 83
60.942
59.291
18.0
CMDS1: ['N170', 'Y110.091']
gcode1 83
60.942
84.691
18.0
CMDS1: ['N180', 'Y135.491']
gcode1 83
60.942
110.091
18.0
CMDS1: ['N190', 'Y160.891']
gcode1 83
60.942
135.491
18.0
CMDS1: ['N200', 'Y186.291']
gcode1 83
60.942
160.891
18.0
CMDS1: ['N210', 'Y211.691']
gcode1 83
60.942
186.291
18.0
CMDS1: ['N220', 'Y237.091']
gcode1 83
60.942
211.691
18.0
CMDS1: ['N230', 'Y262.491']
gcode1 83
60.942
237.091
18.0
CMDS1: ['N240', 'Y287.891']
gcode1 83
60.942
262.491
18.0
CMDS1: ['N250', 'Y313.291']
gcode1 83
60.942
287.891
18.0
CMDS1: ['N260', 'Y338.691']
gcode1 83
60.942
313.291
18.0
CMDS1: ['N270', 'Y364.091']
gcode1 83
60.942
338.691
18.0
CMDS1: ['N280', 'Y389.491']
gcode1 83
60.942
364.091
18.0
CMDS1: ['N290', 'Y414.891']
gcode1 83
60.942
389.491
18.0
CMDS1: ['N300', 'Y440.291']
gcode1 83
60.942
414.891
18.0
.....

vlachoudis, 2015-10-07:
@antoniosap I've pushed a modification that is using parts of your code to make the expansion of the canned cycles. I've tested it against what I found on linuxcnc and it is ok, but could you test it with your inputs.

antoniosap, 2015-10-09:
ok. testing pass for G83.
this is a benefit for community

closed 2015-10-09 by antoniosap