vlachoudis/bCNC #111: Cant change work area

TilTisback, 2015-09-26:
Hi,
i cant change the work area. I have set the travel_x,y,z in the bcnc.ini file but the program seems to ignore it and sets it to 200x200x200 whatever value i type in the ini.
Am i doing it wrong or what value sets the work area? Also changing the max-travel values in the grbl doesn't seem to have any effect.
Pls help, my work area is very small and every time i run into the limit switches.

Effer, 2015-09-27:
Hi @TilTisback ,
are you editing the .ini file in the bCNC folder? If so, that's the wrong file. You should edit a ***.bCNC*** file created in your $HOME directory.
Even if the main page of bCNC is reporting that a graphical setup is missing that's not completly true. You can edit most of settings in the  ***"Tools"*** tabs using the ***"Machine"***  button. Settings are stored in you user setting file, after you close bCNC.

[image]

Please report back if this doesn't help.

vlachoudis, 2015-09-27:
@TilTisback please pay attention that the settings of bCNC are not the same with the settings on grbl, which you set with the $nnn= variables

carlosgs, 2015-09-28:
This isn't working for me either, setting Travel {x,y,z} from the GUI has no effect.  
Was going to propose, as well, to give support to us people that has the origin configured in the front left corner in GRBL. This can be achieved by simply changing the meaning of Travel {x,y,z} to "limits" (values in the negative space would be the GRBL preferred approach -XY=0 back-right corner-, and values in the positive space would give support to XY=0 in the front-left corner). What do you think? How is your set-up?

carlosgs, 2015-09-28:
Correction, the values do get updated, but after restarting bCNC. Negative values work OK (haven't tried running the machine, but I believe the limits are just used as a visualization). I find it counter-intuitive that negative values fall in the positive coordinate space though:
[image]

TilTisback, 2015-09-29:
its working for me now too.
thank you.

vlachoudis, 2015-09-29:
Indeed when changing most of the values a restart of the program is needed.
@carlosgs apparently for good reasons in the past machinists chose that the machine coordinates is on the negative space. There are several discussions on on the grbl why. Just be careful that by setting negative the travel coords bCNC will not think the same way as grbl.

carlosgs, 2015-09-29:
Hmmm. I understand that, thankfully it seems we can choose our preferred coordinate system (I believe the positive one is more intuitive for people coming from the 3D printing background).  

Do you mind if I briefly share with you my workflow? I want to know if it is the correct approach.  
[image]  

Let's say we want to mill a bowl, I'd first move the gantry to the **desired initial coordinates**:
[image]  

Pressing **X=0 and Y=0** centers the coordinate system in the desired spot:
[image]  

Then it is possible to **load/generate the gcode in the chosen spot**. Maybe align it with the editor at this point.
[image]

Finally, I'd **zero the tool length** using the manual tool probe functionality, and **run the gcode**.
[image]

Please let me know if you see any problems with this procedure, I want to share it with some friends but want to make sure it is the correct way to use bCNC! thanks!! :D

vlachoudis, 2015-09-29:
@carlosgs this is more or less what I do as well. Maybe a bit reversed on the fact that first I load the model and then I zero my machine. Sometimes on complicated models I prefer to move the origin of the model to the low-left corner (from the editor toolbar) just to make it easier to align on the machine. And I have the user macro scan that I run on a safe height to see if I might hit something clamps etc...
However don't take for granted my approach I also new to CNC :)

carlosgs, 2015-09-29:
Thank you! I needed this confirmation. Yes, the scan function helps a lot indeed. Must say bCNC is very intuitive, I love it! :)
You can close this issue, maybe consider adding a bubble with the message "Please restart bCNC to apply changes"?

vlachoudis, 2015-09-30:
Y

closed 2015-09-30 by vlachoudis