vlachoudis/bCNC #114: Manual Tool Change

vlachoudis, 2015-09-28:
Hi all,

I would like to get your opinion on the manual tool change approach. The tool change M6 as it implemented right now provides the following choices:
1. Send as is to grbl/controller
2. Remove the M6 command from the line send to the controller
3. Expand with a macro

The present macro expansion does the following
$g -  to remember the state
m5 - stop spindle
%wait  - wait until all commands are processed by grbl and the state is idle
%_x,_y,_z = wx,wy,wz   remember last work position in _x,_y,_z variables
g53 g0 - absolute/machine move to the tool change position first on z and then x,y
%wait - for synchronization
%pause - put on feedhold the machine and display a message on the screen (the comments on the same line of the M6) command. 

g53 g0 - absolute/machine move to tool probing location first x,y and then z
g91 g38.2 - relative probe over the requested length
g10l20... - set the Z coordinate of the current workspace to the correct probe height

which is followed by the invert path change z -> change x,y
stored location in _x,_y and then _z
restore feed and spindle
and continue....

Since I am not at all expert on CNC I have the following issues that I like your opinion:
1. %pause I could replace the feed hold with a M0 and avoid the wait. But I have to find a new way of displaying a message to the gui. Is this the correct use of M0?

2. g10l20... as @chamnit pointed out is that is not conformal with the TLO (tool offset).
Initially I wanted to set the TLO but I had the following conceptual problems
a. TLO is not persistent on reboots
b. TLO would always be relative to the first tool, when the calibration was performed. Which is an arbitrary often not repeatable setting.

To solve (a) would require either to run a calibration after every reboot of the controller (e.g. clicking on stop)
or
let the GUI to set silently the last value it remembered after every reboot.
or
let the user to set it manually pressing a button.

I don't like non of the above options,  calibration would require an additional setting time, setting it automatically or manually could lead of possible errors when we forget about it and change the tool.

My present solution of re-adjusting the Z of the current WCS seemed to me persistent on reboots (but not so elegant) since every time the tool coordinate system is relative to the present setup.

Do you have any comments, what can be the proper way of performing it?

chamnit, 2015-09-28:
@vlachoudis : I had an in-depth conversation with the head machinist at my work about this. Just to make sure that my thinking was correct.

First, WCS must be independent of tool changes and offsets. If it's not, you can lose part zero when you do a tool change, as you have no idea what the previous offset was or what the new tool is relative to.

Second, regardless of what tool change methodology you do, you have to have a reference point. In industry, this is performed by a fixed tool probe location that has a known offset from machine zero. At my workplace, they chuck up a reference tool holder with a precision rod in it. They probe the tip of this reference tool and store the machine position as the initial reference. Then all of the following tools are re-probed relative to the reference. All of the offsets are stored in the machine (or in the case of bCNC in the GUI). 

So, when a job is being setup, the machine needs to know what tool is in the spindle. It'll apply the stored tool length offsets (TLO) to machine space. You find part zero with that tool and TLO active, such that, if and when you change a tool, the new TLO maintains part zero.

The downside of this methodology are that you need expensive tool holders for each tool, a fixed tool probe location, and a tool offset database. This is to speed up production, so you can simply setup all of the tool offsets once and then run the same job, one after another, without re-probing the tool all of the time. (Occasionally you do have to account for tool wear though.) However, this isn't in the realm of noob hobbyist, typically. (But this is very common with home shop machinists or LinuxCNC folks).

To make it simpler for users that don't have tool holders but do have a fixed tool probe location, this is how I would do it (there are tons of other ways to do this):
- First, have the user insert the first tool/edge finder before locating part zero, because you need to measure where the tip of the tool is in machine space before doing anything.
- Set this tip location as TLO zero and store this within bCNC. Also send Grbl the G43.1 TLO offset to apply it and ensure it remove any previous TLO.
- Locate part zero with the first tool/edge finder.
- Apply G10 WCS offsets to set part origin.
- Start the job.
- If an edge finder was used or is a different tool number is specified by the g-code program, the user has to insert the new tool and a tool probe cycle has to be performed. bCNC stores the offset **relative** the initial tool that set part zero. Send Grbl the new G43.1 TLO.
- Upon another tool change, repeat the process above. Install the new tool, probe, and apply TLO relative to the first tool that set part zero.
- Job done. Rinse and repeat for every job.

Without a fixed tool probe location, you just need to have the user input the x,y location of it and require that the z offset doesn't change between probes. For example, this would be common with a touch plate system, where it can be placed anywhere in the machine. Upon a tool change, the touch plate can be put back into place. The user would have to make sure that there is chips underneath that will alter the probe.

I understand that this is all tedious and time consuming, but this is the way it's done. You have to maintain an initial offset such that the WCS is maintain regardless of a tool change. This makes things much easier if something goes wrong with a job. I hope this makes it clearer on how it should be implemented and why professional shops opt for tool holders and changers.

chamnit, 2015-09-28:
Also, if you're concerned about maintaining offsets after a power cycle, you will still be ok with this procedure as long as bCNC recorded the initial TLO that set part zero of the WCS. (It would be possible to have Grbl store it for bCNC in another WCS, like G59, as well.)

vlachoudis, 2015-09-29:
Thanks @chamnit for the detailed explanations! I fully understand the approach setting the TLO, which makes perfectly sense when you have a probing tool well calibrated. In this case the TLO represents the true offset of the tool vs the machine coordinate space.

In my case (but maybe it is just me), which was my first touch-n-go with the tool changing,  I don't have a calibrated probing tool. Every time I insert a new tool in the chunk I cannot reproduce the previous offset. Therefore the whole meaning of setting a TLO versus an initial tool offset, whose memory is lost is practically meaningless. I will be forced perform a calibration on power-cycle or depend on the GUI that has correctly recorded the last TLO (In case of a crash everything will be lost). I see it a bit like using the G92 for setting the work coordinates vs the G54..G59. 
By setting the WCS each time the work coordinates will reflect the actual tool...
BTW I just saw that linuxcnc has a G10 L1 xxx command to save the TLO. Could that be possible in grbl or it is out of eeprom memory on the AT328?

On the other hand my approach is just a confusion for people having tool holders :) or are used to work with the TLO. For this reason I've just created an additional toolChange version that is setting the TLO as you suggested instead of the workspace. I've added it as a different tool change policy that the user can select. I have to test it on my machine before releasing it.

Thinking loudly I should add in the future an extra policy for the users having a tool holder. It could pre-record or manually set the tool offsets, so (s)he is not forced to do a calibration on every tool change.

chamnit, 2015-09-29:
@vlachoudis : I'm not sure I understand what you mean by a calibrated probing tool. All you need is a touch plate that doesn't move and provides a consistent z-axis reading. But I do understand what you mean by you cannot reproduce the tool offset when you chuck up a new tool. That's a common problem, since most people have a collet-based chuck and share tools between it. 

What I meant by storing the first tool, which you use to set part zero, is you need to remember where part zero is in machine space relative to the spindle/tool. So if you store the tool tip location after that initial tool probe, you now have a reference point from which any other tool's TLO can be computed to maintain part zero, even after a power cycle or if you change tools. Just as long as neither the part nor the probe touch plate has moved, you're good to go.  

All the GUI has to do is, upon any M6 tool change it detect, re-probe the new tool to determine the tip location in machine space. Then compute the z-difference between that and the stored reference position that determined part zero. Finally send a G43.1 TLO. It's a fairly straight-forward way to do it IMHO and keeps part zero untouched. 

As for G10 L1, it's not EEPROM that's the problem on the 328. It's flash space for the code to handle new commands and data management. In a Mega/ARM version, we can do integrate tool databases and such, if it seems like the right thing to do. But, I don't think it's a problem to use G57-G59 work coordinate systems for what you need. Most users use only one work coordinate system (usually G54).

onekk, 2015-09-30:
Hi to all,

I have an eShapeoko with grbl 0.9j and bCNC on it, but this is not a problem, I have this:
http://cnc-plus.de/images/product_images/info_images/werkzeuglaengentaster_0811.jpg
But essentially is a switch that close a contact at a certain height over the zero of my table.

I have fixed it near the Home position, so it is relatively  fast to changing the tool (unscrew the collet, place the new tool and screw the collet), and re zero the machine (The machine lose the zero during such manipulations)

Conceptually is the same of having a fixed touch plate,
my probe close the contact at 30mm height
the touch plate at 2mm height.

Some number, when i do:
1) I change the tool
2) I home the machine 0,0,0
3) g38.2 - i obtain a set coordinates, say X 12, Y 25, Z-20 but the only relevant is Z-20
4) bCNC knows that my tooltip is at 30mm above my zero
5) bCNC can calculate that the zero of the workplane is now 30mm lower than Z-20
6) bCNC start the cut knowing where the tooltip is  

When i change tool the new z where the probe close the contact is Z-22
 bCNC knows that the new Zero is 30mm lower than Z-22 and behave accordingly

You have to enter only the Z-height point, once. (maybe you have to verify it from time to time)

If bCNC crash in the middle of a job do 1 to 6  the Z height is always fixed (mine is bolted to the table, but a touch plate maybe could be glued in a corner, maybe under the home position).

Best Regards

vlachoudis, 2015-09-30:
@chamnit with "calibrated probing tool" I've meant as @onekk said out the "reference tool"... I didn't find the correct term :)
I've implemented both methods now, but I have not test it yet on my machine. I will do in the next days I have some PCBs to make, and then I will push it on the master branch.

vlachoudis, 2015-10-01:
I've found some time to go on my machine and test it. Now both methods works ok. When selecting the update of the TLO the TLO is saved on the bCNC ini together with the machine-z of the probing plate/button as it was found using the "reference tool".
However on reset the tlo is not automatically set. The user needs to click the "set" button in the probe page.

closed 2015-10-10 by vlachoudis

arthurwolf, 2016-11-14:
Hey !

What is the status of this ? Is this implemented, is there any way currently to get this to work ?
I am ready to pay whomever is ready to add this, if not.

Cheers !

vlachoudis, 2016-11-14:
@arthurwolf I am not sure I understand what you mean? Manual to changes are implemented, as well what you mention about pausing.

arthurwolf, 2016-11-14:
If they are implemented can you point me at the documentation of it ? I
have not found it.

vlachoudis, 2016-11-15:
@arthurwolf This is a good point, documentation is lacking in many things :)
The tool change functionality is described briefly here https://github.com/vlachoudis/bCNC/wiki/Tool-Change
The pausing can be achieved in grbl with m0 or !
 and syncing to wait until the buffer is empty
 is not documented anywhere the command is bCNC commands
%wait

arthurwolf, 2016-11-28:
Thanks a lot ! I think I can get it to work with that info :) Neat :)

I know I probably already asked you about paying you for bCNC features, but
I wanted to ask you about something else.
We are working on a control interface called fabrica ( web-based ) :
https://docs.google.com/document/d/1YaLVh9GDuIGhbxlQywRCvL1VG922Lr1ZFq7Fv_0r9Cs/edit#
As part of that, we will ( a bit later on, a few months ) be adding a lot
of features that are similar to what bCNC has ( but less "control panel",
more "wizard/step by step" ). Would you be interested in getting paid to
implement those ?

Cheers.

vlachoudis, 2016-11-29:
@arthurwolf Thank you for the offer, unfortunately I don't have the time to work on that.

arthurwolf, 2016-11-29:
Hey, I had to ask, talent isn't easy to find :) Thanks, and tell me if your
situation changes !