vlachoudis/bCNC #115: Mouse Scroll on non Linux Platform

Effer, 2015-09-29:
Hi,
 it seems like that tkinter binds differently the mouse wheel scroll from os to os. This is the reason that zoom with mouse wheel doesn't work on Windows (and OSX I suppose).
I found this cross platform implementation
http://code.activestate.com/recipes/578894-mousewheel-multiplatform-tkinter/
we could use as reference for fix the issue.  
Also to report that, on Windows and with Italian Keyboard, the Ctrl-= ("zoom in" shortcut) is not working. Ctrl-- is instead working correctly. 
This probably depends on fact that in Italian keyboard the symbol '=' needs also the shift button to be pressed.
Settings the keyboard to English (and using the correct keys) make both shortcuts works again correctly.
For the one with NumKeyPad could be enough to add also Ctrl-+ as shortcut for "zoom in".

vlachoudis, 2015-09-29:
I believe that I fixed that (at least for windows) can you check it. For mac I am not sure. After several months I rebooted my laptop in Windows and tried for the first time bCNC on windows :)
Now I've modified the program so when it starts on windows and the serial is missing it issues also a message on how to install it.

Effer, 2015-09-29:
Confirm, it works. 
Good job @vlachoudis

closed 2015-09-29 by Effer