vlachoudis/bCNC #122: Network file support

atrueresistance, 2015-10-07:
Would it be possible to have network drive support? 

Currently the open menu looks like this:
[image]

I could be just a dumb user, but I can't seem to get it to change from C: drive. Currently I have my nas attached to the Z: drive. I just can't figure out how to traverse in bCNC. Otherwise great software!

Effer, 2015-10-07:
Hi @atrueresistance .
I've just made this test in Windows: 
mount a remote disc with net use in Z:

[image]

then input z: in the "File Name" text box

[image]

Indeed not so user friendly, a list of local disk would be useful.

atrueresistance, 2015-10-07:
Awesome thanks, silly user error on my part. I kept clicking on the C: next to the directory thinking there would be a dropdown to select another drive.

Effer, 2015-10-07:
@atrueresistance to be honest, was exactly what I've done the first time ;)

vlachoudis, 2015-10-07:
I've found on the web that I can get the logical drives with the win32api.
if it works i could add it in the dialog box to propose them

import win32api
drives = win32api.GetLogicalDriveStrings()
drives = drives.split('\000')[:-1]
print drives

Effer, 2015-10-07:
It works but it seems that win32api is not installed as default.
Dependandy for dependancy, what about psutil?

import psutil
psutil.disk_partitions()

Anyway it's a minor issue IMO.

closed 2015-10-20 by atrueresistance