vlachoudis/bCNC #123: smoothie connection

steiereh, 2015-10-08:
loaded bCNC and looks great. op. running on raspberry pi 2 open connection but no connection label does not change and does not control machine

steiereh, 2015-10-09:
What I mean is the no connection window does not change. if i click on it, it says connected to grbl but does control machine. nothing happening. I am trying to use cnc machine for route, laser cutting, printing. Pronterface connects the smootie board and jogs it. Now trying to bCNC to connect and operate smoothieboard. nThanks for any time to help.

vlachoudis, 2015-10-09:
Is your machine in lock state. If yes you need to press the unlock button first.
Check on the terminal tab.

steiereh, 2015-10-09:
if i click unlock under control nothings happens. looking at terminal it shows
smoothie
ok
$X
commands
26 lines later 
calctermistor
does this help you?

steiereh, 2015-10-09:
[image]
[image]
[image]

vlachoudis, 2015-10-10:
Sorry @steiereh I didn't know that smoothie is a different controller. bCNC is made to work with grbl https://github.com/grbl/grbl only. At least for the moment.

steiereh, 2015-10-10:
I am confused this what smoothie says about there controller on git hub. Your saying smoothie is not grbl? 
Smoothie is a free, opensource, high performance and modular G-code interpreter and CNC controller for the LPC17xx micro-controller ( ARM Cortex M3 architecture ). The motion control part is a port of the awesome grbl. The Source code is on GitHub 
Thanks fir your time is this matter. I am trying to learn about python and linux and cnc machines

vlachoudis, 2015-10-10:
I don't know what smoothie is, but what I see in the terminal output has nothing to do with how grbl is communicating. The present version of grbl runs on arduino (atmel atmega cpus), not ARM. I am sorry, but as it is right now bCNC cannot communicate with smoothie.

closed 2015-10-10 by vlachoudis

steiereh, 2015-10-10:
No problem i will do more research on smoothie.