vlachoudis/bCNC #125: Can't add fields in the Material and End Mill databases

justinclift, 2015-10-09:
Finally getting around to trying out bCNC properly.  Went to add some materials and End Mills... and there doesn't seem to be any way of adding required fields.

For example, in the End Mills database it has a "Length" field already there (is that flute length, overall length, unsupported length, or something else?).  I'd like to make sure all the End Mills can have at least flute length and and unsupported length, as they directly affect feed speed and usable rpms.

For materials, it has Feed, Plunge Feed, and Depth Increment... but the figures should be are specific to an end mill/machine configuration and not generic.  eg milling (say) 6082 T6 aluminium at 26mm depth, 0.9mm cutting edge, 522mm/min feed rate, is specific to a 10mm HSS tool bit with unsupported length of 50mm.  If the end mill differs, so do the feed settings.

So... being able to add fields to the End Mills database would help.  The Materials one... seems like it needs a different approach. (I could be wrong though, and not seeing something obvious :wink:)

vlachoudis, 2015-10-10:
@justinclift I didn't foresee in the interface to add/modify the database fields.However it can be done if you modify the ToolsPage.py on the relevant tool. You are welcome to do so and make a pull request.

Concerning the materials you are perfectly right the feed rate should match the material, depth of cut, and end mill properties. I am open to suggestions on how to implement it, what are the necessary parameters and how to calculate the feed rate. I saw somewhere some formulas on the surface speed.

justinclift, 2015-10-10:
Make all makes sense.  I'll take a look at ToolsPage.py in a bit, and see how I go. :wink:

justinclift, 2015-10-14:
As a related data point, the Wikipedia page for "speeds and Feeds" has useful equations:

&nbsp; &nbsp; https://en.wikipedia.org/wiki/Speeds_and_feeds#Spindle_speed_calculations

vlachoudis, 2015-10-15:
Thanks for the pointer. The only thing that is not clear for me, is what are the necessary parameters to have in the database. My approach is a bit naive, but I found it easier to control.

More correct would be to have a field to insert (meters per min) or (surface feed per min) and let the program to calculate the feed rate, based on the depth of cut, spindle rpm, bit diameter and flutes. If this is the correct approach I believe then I would need to have a feed rate calculator to do also the invert calculation for the user. Once I know that some settings works for my machine, I plug them and the program calculates the MPM or SFM.

justinclift, 2015-10-17:
Good thinking.  We could probably use a source of speeds/feeds data like this:

&nbsp; &nbsp; http://www.drill-service.co.uk/images_Technical/Speeds-and-Feeds.pdf

... and with the above formulas we could calculate a good range of suggested settings to meet the user's target.  Not quite sure exactly how yet (I still haven't gotten around to looking at the source). :innocent: