vlachoudis/bCNC #127: Saving probe data

pommen, 2015-10-11:
Hi,

First off: Awesome piece of software, I use it all the time for PCB milling.
I updated yesterday to lastest version and cant see where to save the probe data.

Usually i probe the whole PCB, save the probedata and then mill the text layer. Then I load the top side file and load the probedata and continue milling.

Is the functionallity to save the probefile gone, or am I searching in the wrong places?

Kepp up the good work!

/Peter

vlachoudis, 2015-10-11:
In the effort to reduce buttons in the interface the saving/loading of the probe data is done with the same save/load button for the gcode. Simply change the extension to probe

pommen, 2015-10-11:
Ahh I see, thanks!

/Peter

closed 2015-10-11 by vlachoudis