vlachoudis/bCNC #130: Machine space orientation

jdiderik, 2015-10-20:
Complements to start with, love bCNC.

I went through the code and could not find a setting to make the machine space work from bottom-left, thus making it positive on x and y.
I read in the wiki that this is something left over, is there anyway to change this?

I have my cnc setup (with a pi2 as bCNC host) having home (0,0) bottom left. The canvas painting the workspace is always incorrect and machine coordinates are max negative values when homed.

I would love to have this work differently, especially since this is easier to explain to other users of my machine.

carlosgs, 2015-10-20:
You can set the XY travel to negative values, that will do it!

A bit counter-intuitive, but I use it with Cyclone PCB with no issues at
all.
Cheers,
Carlos

jdiderik, 2015-10-20:
Might work, will need to try when back behind machine. But I think a combination of the grbl setting $23 (home dir invert mask) with $130 en $131 (max travel x and y) should do this automagically to make it easier on beginning users.

vlachoudis, 2015-10-20:
I was confused as well in the beginning when I realized the the machine coordinates are in the negative space :) Its a typical confusion, due to tradition in the CNC world, that negative goes towards the object and positive goes out.
However once you home your machine everything should show up ok. Unless if you don't have homing switches.

carlosgs, 2015-10-20:
Honestly I'd rather not have to learn another coordinate system method, I prefer to stay with the same one 3D printers use. Thankfully this is possible by having a CNC with min-endstops and enabling the following flag in GRBL:

```
// After homing, Grbl will set by default the entire machine space into negative space, as is typical
// for professional CNC machines, regardless of where the limit switches are located. Uncomment this 
// define to force Grbl to always set the machine origin at the homed location despite switch orientation.
#define HOMING_FORCE_SET_ORIGIN // Uncomment to enable.
```

Having worked with 3D printers for such a long time, it is very relieving to be able to work with the same coordinate system (definitely easier to visualize for me).

I agree with @jdiderik in that it would be nice to have bCNC auto-detect these configurations, but in my view it is not a priority. Truly appreciate the fact that bCNC works seamlessly with both GRBL configurations :-)

vlachoudis, 2015-10-20:
Initially I wanted to autodetect most of the parameters like travel distance, acceleration etc. but since the setup changed from 0.8 to 0.9. I though that the effort to make it automatic is bigger than just entering a few variables again in bCNC setup :)

Effer, 2015-10-20:
Even for me would be nice to auto detect parameters. Is what most of user expect, and often is cause of confusion on how change the correct ini file.
If grbl 1.0 should keep the same parameters, it worth the price to autoimport them.

Guenni75, 2016-05-09:
Thumbs up for autodetect 😁

chamnit, 2016-05-09:
I think I shed some light on this coordinate system issue. 

On a 3d printer, you have a fixed extruder and start a print from build plate and print upwards Z+. It completely makes sense to place the origin in the left-front-bottom corner, such that all of the distances are in positive space. 

With CNC subtractive machines, it's the opposite. You have piece of stock in the machine and you remove material from the top of the stock, on downwards Z-. Also, since end mills and tools have varying lengths, you don't have a repeatable reference point from the bottom, like you do with a 3d printer. The logical place to place the coordinate system is the rear-right-top corner of the machine, such that you reference the z-travel, rather the tip of the tool, and all of the values are consistently negative. Otherwise, you'll have a few negative and others positive. 

The CNC coordinate frame is also historical in nature. It's always been this way, and 3d printers have just recently moved the coordinate frame to the front-left-bottom. 

In the end, where the coordinate frame does not matter. CAM programs almost NEVER use the machine coordinate system. It's very dangerous to do so. They use work coordinate systems. Once users learn how to use work coordinate systems, machine coordinates have no real meaning.

chamnit, 2016-05-09:
@Effer: FYI, in v1.0, I'm planning on providing some kind of build code that indicates what config options are enabled and other important details that may be important for a GUI, like RX buffer size, home origin location, etc. It'll have to be very short and compact, so it'll probably involve some values/characters that need to be looked up on a table (that I'll provide in an API document.)

Effer, 2016-05-09:
@chamnit thanks for the nice clarification about the coordinate system. When [WCS Page](https://github.com/vlachoudis/bCNC/wiki/WorkSpaces) will be updated they will fit well as general introduction.

About the config options: we'll implements them quickly as soon V1.0 will be aviable. Keep us updated.

closed 2018-10-29 by Harvie