vlachoudis/bCNC #131: Toolchange with manual spindle control

jdiderik, 2015-10-20:
The toolchange sequence works great!

I do have one issue with it in my workflow / machine setup:
- When a T[T]M6 is encountered, it nicely moves to my toolchange position
- It waits for me to change tool, during which I can turn off the spindle, change tool and attach the probe clamb to the toolbit (manual, DeWalt 611)
- Hit resume to start probing sequence
- But after probing it will continue the job straight away, but my spindle is still off and my probe-clamb is still on my bit... you can imagine the mess it will make if I let it continue... :-)

Could there be a .ini setting to define that we have manual spindle control / probe-clamb and when set that it will add an extra FEED_HOLD / PAUSE into the probe/toolchange sequence after probing (with a comment)? Maybe even have the status say something like: 'Remove probe / Start spindle' in bCNC.

vlachoudis, 2015-10-20:
Ooops, Since I've upgraded my spindle to PWM controlled by grbl @jdiderik I completly forgot for manual spindle control :)
I've added an option in the Tool Change page whether to pause the machine at the tool change position after the probing. (This is the default for safety now).
However I didn't test it yet. Could you test it please.

vlachoudis, 2015-10-20:
I've just tested it. Now it is ok.

jdiderik, 2015-10-22:
Hi @vlachoudis,
Tested and works great! Thanks for the quick fix.
I tried changes to the code myself, but I'm not that fluent in Python.

vlachoudis, 2015-10-24:
You are welcome

closed 2015-10-24 by vlachoudis