vlachoudis/bCNC #135: Python27 error

Tuga75, 2015-10-21:
I have bcnc running on windows 7 i have python27 install and pyserial. i´m using ramp1.4 on cyclops pcb and the machine runs sometimes ok and order time bcnc frezes and program stop responding an i get this error i try other pc and the same thing hapens any one have a solution.

Best regards
Paulo
[image]

vlachoudis, 2015-10-22:
This error messages point to a serial communication error. For some reason the line is dropped or never connected.

Effer, 2015-10-22:
Hi @Tuga75
 is the error always the same? I mean, is always happening when bCNC is sending a status request:
self.serial.write(b"?")

Which grbl firmware are you using? You can see from terminal Tab during the boot of the controller.
Anyway a poor usb shielded cable can bring a lot of issue.

Tuga75, 2015-10-22:
Tanks for the reply the grbl as from cyclone pcb factory https://github.com/carlosgs/grblForCyclone to use on mega2560 + ramps 1.4 , yes the terminal is present on lanch bcnc i am trying another usb cable.

Paulo

closed 2015-10-27 by vlachoudis