vlachoudis/bCNC #139: Profile tool does not respect/use stock thickness and material step down?

mandrav, 2015-10-24:
I wanted to create a profile for a path and I have correctly setup my endmill, material and stock in bCNC.
But no matter what I try, it always generates a single path on Z=0 (my surface Z in stock).
What I would expect is to repeat the path in material-depth-increment steps until stock-thickness is reached.
Or do I not use the tool correctly?

PS: I think the same goes also for the pocket tool.

carlosgs, 2015-10-24:
I had the exact same issue yesterday; it turns out what you need to use is the "cut" tool after running a pocket/profile operation :-)

mandrav, 2015-10-24:
Ahh! Thanks Carlos, will check shortly :)

vlachoudis, 2015-10-24:
Indeed. The profile and pocket only generates the path, then you have to apply a cut.
Maybe I should combine it in one.

mandrav, 2015-10-25:
Confirmed!
A bit counter-intuitive at first but works fine :D

One thing I noted is that the cut provides its own surfaceZ and step-down values, ignoring the respective values from stock and material. I like the fact that it allows me to verify the values in one place before producing the cut but, maybe, it should also read the values from stock and material to use as a starting point?

@vlachoudis great job! This program always has something up its sleeve for me ;)

vlachoudis, 2015-10-25:
When I first implement the profile/cut I didn't know what parameters to put, so I've split it in two. I believe that I could now safely add the cut parameters also in the profile, pocket etc.

In the "cut" tool what ever value you leave empty (not zero) it will use the one from the stock material.

closed 2015-10-26 by vlachoudis