vlachoudis/bCNC #140: Drill position calibration

yoyokal, 2015-10-25:
Because my wife doesnt like the noisy sound my cnc when milling pcb, I just use my cnc to drill it. I'm using transfer toner to create track. The problem is my laser printer doesnt have the same acuracy with my cnc. I know I could change the value of mm/rot on grbl, but its not elegant, because I must guessing the value. 

To solve this issue I hope bCNC have a function to calibrate drill position. Im propose the procedure as follows.

1. Set home x0, y0.
2. Click Calibrate Drill button. bCNC will travel to very bottom left corner drill hole and stop.
3. User calibrate the position of the first hole using controll drill manually. After the drill spot on correct hole a button clicked.
4. The drill the move to very right-top corner of hole and stop.
5. User calibrate again as step 3 and calibration is finished.

With this procedure we could drill pcb created by other machine without worries the varies accuracy with our machine.

What must solve is finding the formula to compute real coordinate because the difference value is dependent to position. I mean the difference value on 1mm will different at 10mm, maybe the function is logarithmic.

I'm sorry if my English isn't good. But I hope my issue and my hope are fully understand.

vlachoudis, 2015-10-25:
Hi @yoyokal. I am bit astonished that the laser printer doesn't produce the same accuracy with the CNC. If the machine is well calibrated it should match the print out, unless if your machine is not squared. If we assume that the machine is not properly calibrated and not squared we can find the transformation matrix between your pcb  reference system and the machine with minimum 3 points, the more the better would become. This is the aim of the "Square" tab inside the "Probe" page. For the moment is not implemented, but in principle is easy to do.

yoyokal, 2015-10-25:
Ah... I didnt see this function... really I'm bit strugle with user manual.
Thank you, I'll try to do it.

vlachoudis, 2015-10-25:
Maybe I was not clear, the function is not implemented yet. I will do it when I have some spare time.

yoyokal, 2015-10-25:
Oh, Im sorry if I'm missunderstood, thats why I wonder my square menu is
greyed.

Effer, 2015-11-17:
@vlachoudis  do you think that square implementation could be used as base to add a scale command? Scale is really missing from editor.

vlachoudis, 2015-11-18:
Scale would indeed be helpful. As well rotation around a pivot point instead of the origin.

I have the routines ready that given a set of x,y/real <-> x,y/measured points to calculate
the transformation matrix, so one can rotate, translate and scale the gcode to fit the table.

Effer, 2015-11-18:
That would be great!

Filippo

----- Messaggio originale -----

vlachoudis, 2015-11-18:
I am not sure on how to integrate on the interface, and how to properly probe the x,y points.
1. I was thinking using the center probing ring to probe one by one several points. e.g. clicking on the canvas and then probe it. But I am not fully convinced.
2. Another approach maybe easier is to stick two L-metal bars on two perpendicular edges of the piece to align and probe with linear x or y motions the edges...

Any good idea?

Effer, 2015-11-18:
I like the first one. What doesn't convince you? 
Select first point in the canvas. Grbl move machine to that point at safe height. User adjust with jog the "real" position and then accept it with a probe command or by GUI button. Save position and repeat for at least 3 points. Done, right?

For scale and rotate in the canvas I would be nice to add two more buttons in the Editor ribbon like the Move command.

vlachoudis, 2015-11-20:
@Effer the problem on the first is how to accurately probe the x,y. The only option I see is the probing ring. While on the second option the machine can probe automatically several points which will all be in a straight line

1bigpig, 2015-11-20:
@vlachoudis <https://github.com/vlachoudis>

I have been thinking about the scaling issue.  First off, I know that laser
printers can be notoriously inaccurate as the paper can shrink/stretch
under the heat and pressure of the rollers inside the printer.

Now, I think that you would only have to scale from 4 points.  I do think
that while the ring would be "best", I think that you will be "close
enough" with a sharp pointed tool pointing over a known hole location or
where a hole should be.  At least for through hole circuit boards.

The upside of this that as long as the moisture content of laser printer
medium does not change and you follow the same layup technique, you
"inaccuracy" should be pretty consistent from board to board.

Now, if you are trying to compensate for lead screw wear/drunken
threads...all bets of are off, including the ring probing method.

So, I think that if the user takes two points in the X axis and 2 in the Y
axis, this would be good enough for the scaling feature.  At least from my
understanding of what you are trying to do.

Bruce

vlachoudis, 2015-11-21:
@1bigpig I was simply thinking how to position on the machine an existing piece for further processing. Until now I was making a pocket in a scrap wood and placing the piece inside. To avoid this additional step I was thinking that if I could get a set of x,y points on the sample, then calculate the transformation matrix, rotation, translation, scaling, skewness. Actually I have done this procedure before and I have the algorithm for also higher order corrections but is not necessary. Once I have the transformation matrix, and then I can transform the gcode to match the piece.

By solving the over-determined system, the more points you provide the higher accuracy you can get, but when you want to make holes on a pcb you need sub-mm precision.

vlachoudis, 2016-02-04:
@yoyokal, @1bigpig  I've just uploaded a new version of bCNC, that is able to orient (align) a working piece gcode vs the real coordinates in the machine. Actually it will transform (rotate/translate) your gcode to match the piece mounted.

There is no documentation for the moment, but here is a brief explanation of how it works.
1. Mount your piece on the machine. e.g. your PCB that you want to drill the holes only (as you have etched it chemically)
2. Load the gcode
3. Go to the tab "Probe" sub-tab "Probe". In the present version the simple Probe, Center Probe and Orient are all there, under the same tab.
4. You need to add a few markers.
- First jog the machine to the position you want to mark
- Click on the "Add" button from Orient and click with the mouse the location on Gcode that corresponds to what you have moved the machine to
- Repeat to add a few markers. Min 2, but the more you put the better the accuracy will get
- On every marker added the rotation and translation will be displayed below
- If you make an error delete the marker or manually edit the values
5. Once you are happy with the markers and the error displayed, select the gcode you want to align, or don't select anything (it will take the whole) code and click the "Orient" button
You will see that your code is transformed to match the orientation points.

If everything works you can mill happily ever after.

Note:
- you can save/reload the orientation points, using the extension ".orient"
- I have not tested it on a real piece, but virtually. Please report if it works correctly.

Effer, 2016-02-04:
That's awesome @vlachoudis !
Sad I can't try it now.

lalo-uy, 2016-02-05:
Loved it, will try it today and inform you.

vlachoudis, 2016-02-07:
Today I've test it in one piece. It worked nicely! I had to change the machine coordinates with the work coordinates it makes more sense, and also to draw the errors relative to each marker with a circle. It helps to correct the problematic ones.
The more marker/points you give the better accuracy you can achieve. However it is very difficult to position accurately in X-Y the gantry with the eye. I used the positioning ring from @1bigpig and I could get a bit better than 1mm.
I am more thinking of attaching a pinhole camera on a shaft and mount it on the router.

yoyokal, 2016-02-09:
It's great news. I will try it, but sadly I can't do it right now. My daughter is at hospital.

vlachoudis, 2016-03-05:
Just a small update. In my development branch I've added support for an alignment camera next to the spindle. For testing  I am using a cheap low resolution (356x288) web cam modified to focus at ~3cm, I can get accuracy down to 20um! Combined with the orientation algorithm I am now able to precisely align. I will order a microscope camera for better results :)
When I will have a working interface I will push it in the master.

vlachoudis, 2016-03-27:
A very preliminary camera alignment system is now included in the master version. Despite being primitive is very powerful method for alignment, and with the use of python-opencv it opens a lot of possibilities. If I find some time next days I will write in the wiki how it works.
Briefly:
1. install a web camera next to the spindle at a fixed distance. 
2. Adjust the camera focus lens to get a net image at small distances.
3. Additional light is strongly recommended.
4. Turn on the camera from canvas
5. Move the spindle to some test area and make a test hole
6. Mark the spindle position from the "Probe->Camera" tab
7. Jog the gantry to center the hole in the camera. Optionally adjust the crosshair to fit the spindle diameter, and for the moment manually adjust the "scale" to match the camera pixels/units (mm,inch)
8. Mark the camera position
Tha

vlachoudis, 2016-03-27:
Sorry for the open/close I've hit by mistake the close with my touchpad :)
The image scaling I want to do it automatic, but I have to fix some synchronization issues with the thread that sends the commands first

Effer, 2016-03-27:
@vlachoudis holidays are really good for you! New camera alignment is another big improve for bCNC. (and I believe no other software have half the features as bCNC).

vlachoudis, 2016-03-28:
I've added some information on the wiki page on how to align and orient with the camera
https://github.com/vlachoudis/bCNC/wiki/Probe-Camera-Alignment

closed 2016-05-04 by vlachoudis