vlachoudis/bCNC #141: Is Tools->Tabs implemented?

mandrav, 2015-10-25:
When I press it, it actually displays the machine settings...

vlachoudis, 2015-10-25:
Nope. They are not implemented yet. It is a place holder for me to remind me to implement it :)

closed 2015-10-26 by vlachoudis

vlachoudis, 2016-01-30:
@mandrav Now they are implemented and working fine, maybe they still need some adjustment

mandrav, 2016-02-01:
Thanks, will try it out when as soon as I get the chance :)

mandrav, 2016-07-20:
Hey @vlachoudis , care to briefly outline how they 're supposed to work? I can add them to the design but they don't seem to have any effect on it, probably because of my ignorance.

ThierryM, 2016-07-20:
Le 20/07/2016 à 15:38, mandrav a
      écrit :
    
    
      Hey @vlachoudis
        , care to briefly outline how they 're supposed to work? I can
        add them to the design but they don't seem to have any effect on
        it, probably because of my ignorance.
      —
        thread.
          it on GitHub, or mute
          the thread.
      
        
          
          
        
        
      
    
    
    
    Hi,
    I've just tried to use the tabs function but like you, that
      doesn't work because this function just adds the tabs but the
      first track (for me it is a G-code file made under Inkscape) isn't
      modified.
    So if the depth of your first track is 6mm and the tabs 3mm, 6mm
      wins ;-). It's
      like this I understood the problem.
    Regards,
    Thierry

vlachoudis, 2016-07-20:
With the "Tabs" command you place the position of tabs on a path, and when you issue a "Cut" command the new path will respect the tabs

ThierryM, 2016-07-20:
Thanks @vlachoudis for your explanations. I didn't perform the "Cut" command after I've created the tabs. Now the new G-code seems to take the modifications in account. I'll test tomorrow.
I find bCNC more powerfull days after days. Thanks for all.
Regards.