vlachoudis/bCNC #144: on Linux: rtscts should be True

jcheger, 2015-11-02:
First of all, thanks for your wonderful job.

I have a problem on Linux (Mint 17.2 amd64). Once the arduino connected (FTDI, grbl 0.9i), the Sender is muted. Commands can be sent, but without answer. BTW, the status shows disconnected, while clicking on it says connected.

If the connection was opened once with another program (ex. screen), the connection is alright, and shown as connected. This only happens right after a physical connection.

After googleing a while, I found other people having the same issue. Some did solve it by using the command "stty --file=/dev/ttyUSB0 crtscts -hupcl", and are pretending they have to send the command from the python script, in order to have the serial port working. This is what I've done, and it seems working for now.

Sender:368

		import platform
		if (platform.system () == 'Linux'):
			os.system ("stty --file=%s crtscts -hupcl" %device)

vlachoudis, 2015-11-04:
Thank you. It seems that pyserial in your linux doesn't enable the hardware control. I will check if it is possible to avoid the system call and do it from python directly

jcheger, 2015-11-08:
I've made a mistake in my previous comment. This board embeds a CH340 serial convertor, instead of FTDI. This may also make a difference. It seems that all recent chinese and low cost boards use this chip.

onekk, 2015-11-08:
I'm using the CH340G in my board without any issue, but i use an Olimex board running Debian to communicate with GRBL.

strom-peter, 2016-01-13:
I'm got the same problem, with CH340G and ubuntu14.04.

chamnit, 2016-01-13:
FYI. There has been some thorough testing done by some Grbl users showing the CH340G USB-serial chip has problems with streaming a lot of data. Its internal buffer frequently overflows and corrupts. This type of behavior was not seen or greatly reduced in atmega8/16 or FTDI chips on official boards. 

Some solutions have been to alter Grbl's baud rate, increasing it to 250000 or decreasing to 57600. But this has been hit or miss on whether this fixes the problem or not. My recommendation is to avoid CH340G-based Arduinos for Grbl.

strom-peter, 2016-01-13:
I test different baud-rates, always the same. I will lock at another USB-serial chip and test. Bud, with UGS its works fine!

@vlachoudis - Thanks for bCNC!

jcheger, 2016-01-14:
FYI, I did change my computer 2 months ago, and I had to upgrade the kernel to 4.3 for compatibility issues. I did never have this issue again. I still use the same chinese arduino and the same cable.

Another observation, my laptop is running with Mint 17.3, with the official kernel, and I've never got this issue neither. I've striked a new machine up this evening, but it got frozen at the beginning. Afterward, I did run a one hour job without any communication issue.

strom-peter, 2016-01-14:
OK, after change python-serial-2.6 to 3.0 it is working.

On https://github.com/vlachoudis/bCNC/wiki/OS, I found:
     2) To update pyserial, install pip and upgrade pyserial:
         sudo apt-get install python-pip
         sudo pip install pyserial --upgrade

Also python-serial 2.7 in ubuntu 15.04 with 4.3. Kernel is working with my ArduinoClone/CH340G.

@all - Thanks for input

vlachoudis, 2016-01-17:
Thanks it would be good to update on the wiki on the python-serial requirement.

closed 2016-01-17 by vlachoudis