vlachoudis/bCNC #150: Issue with GCode file encoding

danst0, 2015-11-13:
Hello,

I get the following error when opening my GCode file (attached): 
`
Traceback (most recent call last):
  File "/home/pi/bCNC/Utils.py", line 284, in __call__
    return self.func(*args)
  File "/home/pi/bCNC/bCNC.py", line 1625, in loadDialog
    if filename: self.load(filename)
  File "/home/pi/bCNC/bCNC.py", line 1653, in load
    Sender.load(self,filename)
  File "/home/pi/bCNC/Sender.py", line 332, in load
    self._saveConfigFile()
  File "/home/pi/bCNC/Sender.py", line 310, in _saveConfigFile
    Utils.setStr("File", "file",  os.path.basename(filename))
  File "/home/pi/bCNC/Utils.py", line 196, in setStr
    config.set(section, name, str(value))
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe4' in position 21: ordinal not in range(128)
`
It seems to be related to the filename (without the "ä" it works).

[20151016 Netzteil Gehäuse_v5_Bodenplatte.txt](https://github.com/vlachoudis/bCNC/files/33817/20151016.Netzteil.Gehause_v5_Bodenplatte.txt)

Daniel

vlachoudis, 2015-11-13:
Thanks is the saving on the config file that doesn't like the unicode character. I will convert it to UTF-8 when writing & reading

vlachoudis, 2015-11-18:
I am now using UTF for the filenames. So it should work with non-latin characters.

closed 2015-11-18 by vlachoudis