vlachoudis/bCNC #151: autolevel issue

Dorsel89, 2015-11-13:
Hey, 
i have some problems with the autoleveler. i can use the probefunction via the terminal. for example if i type the command G38.2 Z-10 F10 i can measure the height of my table. if i try to use the autolevel function the software dont recognize the voltage on the probe pin. i found out that if i use a really short cable to wire the ground pin with the probe pin it recocgnize a signal and go on to the next point. the strange thing is that i can use this trick to get the first point and after this i can use my long cables and the autolevel function continue with the normal set-up. 
im using bcnc version 0.7.0 and grbl 0.9i. does somebody have an idea what the problem can be?

1bigpig, 2015-11-13:
@Dorsel89 
I had something similar to what you are describing on my Chinese 2030 machine.  I am using the stock controller and I was getting similar intermittent probe signals.  I think there was noise on the line or the Arduino was not able to pull the voltage on A5 high enough after a "touch".  I was able to solve my problem by connecting a 1K Ohm resistor between the +5Volt and the A5 probe port.  This solved my probe problems 100%

Hope this works for you as well!

Dorsel89, 2015-11-13:
thx  i will check this tomorow morning, the strange thing is that i can use the normal probe command in the terminal.. thats the reason why i dont think there is a "physical" problem with the wiring. anyway i will try this

vlachoudis, 2015-11-13:
@Dorsel89 can you check on the terminal window of bCNC if grbl is reporting back the PRB

Dorsel89, 2015-11-14:
hey, its working now! the problem was me :/ i tryied to autolevel from zmin -20 to zmax +20. if i change zmin and zmax everything is fine. anyway, thank you for your fast response!

vlachoudis, 2015-11-14:
Nice!

closed 2015-11-14 by vlachoudis