vlachoudis/bCNC #153: French translation

ThierryM, 2015-11-18:
Hi,
Thanks for your work and sharing.
How could I add a french translation of the GUI or in an other language ? Of course, it's for sharing.
Regards,

Thierry

vlachoudis, 2015-11-18:
I didn't think of translating to other languages :) I saw that can be done using the gettext python module. I have to read how it works and let you know.

ThierryM, 2015-11-18:
Thanks for your reply. I will help with pleasure.

onekk, 2015-11-23:
Me too @vlachoudis, I volunteer for Italian Translation (I've done an Italian Translation and an Italian Guide for CamBam).

vlachoudis, 2015-11-23:
I've added a makefile that generates the .pot needed for the translation.
@ThierryM  I've made for testing the French .po and .mo files using the [poedit](https://poedit.net/) program
@onekk you can create a subdirectory it/LC_MESSAGES and do there the Italian translation.
it should go on the bCNC domain.

I've did quite quickly the change with the _("") infront of the strings that needs to be translated, so I wont be astonished if I missed something, or I've added it somewhere else which I shouldn't.

For the moment you need to set the environment variable LC_ALL to change the language
like
`export LC_ALL=fr_FR`
I will create an option inside the setup to change the language from bCNC directly, and independent from the OS.

ThierryM, 2015-11-23:
Hi,
Thanks for your work. I'm going to glance at the .po file.
Regards

onekk, 2015-11-26:
Hi @vlachoudis I'm translating bCNC but after cloning the dir I've modified (wrongly) the bCNC.pot file, but now I'm trying to regenerate it issuing "make pot" but it won0t work, what I'm doing wrong?

vlachoudis, 2015-11-26:
Do you have the xgettext installed?
Why not revert back the pot from the repository?

onekk, 2015-11-26:
Hi, Yes I have xgettext installed and even poedit i will try to revert it back maybe only the pot file, i will study the git documentation.

vlachoudis, 2015-11-26:
You can open directly from poedit the pot and then save it as  it/LC_MESSAGES/bCNC.po

onekk, 2015-11-27:
Ok I'm working on it

ThierryM, 2015-12-21:
Hi,
I have ended the french translation (it's not perfect but I think it works) with Poedit 1.8.6 on Ubuntu 14.04 (hard to compile it...). So how to integrate the french files bCNC.mo and bCNC.po in bCNC ?
I'm a beginner on Github... Here is the fork : https://github.com/ThierryM/bCNC
Regards.

ThierryM, 2015-12-21:
Hi, I have tested the translation and there are some words not translated (missing in the .po file ?). See the attachments

[image]

[image]

Regards

vlachoudis, 2015-12-21:
Many thanks @ThierryM. I've uploaded the new translation and I've added most of the missing keywords in the pot. The tab names and group names for the moment are not there yet.
Their value is used by the program to dynamically configure the ribbon from the ini.
I need to find a way to leave the english version in the ini file and the translated to appear on the screen.

onekk, 2015-12-23:
@vlachoudis I'm working on the Italian Translation, I've made quite some work, the main problem i foaund is to figure out where the terms belongs to the interface. But i'm working and testing the tranlsation.
I intend to work on in the Christmas Holidays, so maybe for the new Year I make a pull request.

Regards and Merry Christmas to You and to all.

Carlo D.

dodolbysuround, 2016-01-22:
Hi, I'm trying to use the french version of bCNC but don't kow how to configure it?
Could anybody help me? I'm runing it on windows 7 
Thank you (-;

Effer, 2016-01-22:
Hi @dodolbysuround ,
under windows I change manualy line 26 on bCNC.py
from this:
__builtin__._ = gettext.translation('bCNC', 'locale', fallback=True).ugettext

to this:

__builtin__._ = gettext.translation('bCNC', 'locale', fallback=True, languages=['fr']).ugettext

I think @vlachoudis  is going to implements a more user friendly way soon.

vlachoudis, 2016-01-22:
@dodolbysuround  It was a bit tricky to add it from the config since it has to be executed before any fixed string translation takes place. Now I've modified the code so the user can select the language from the "Tools" page

Effer, 2016-01-22:
Well done @vlachoudis 

[image]

Being multilingual will help lot of users.

ThierryM, 2016-01-22:
Thanks @vlachoudis

vlachoudis, 2016-01-22:
Actually the kudos goes to the translators @ThierryM @carlosgs @onekk who did the hard job.

dodolbysuround, 2016-01-22:
Thank you everyone!

vlachoudis, 2016-02-03:
@ThierryM one small correction. The "Vertical" is translated to "Horizontal"

ThierryM, 2016-02-03:
Hi, correction done + french translation updated.
Regards.

tuxun, 2016-03-01:
Hi to everyone!

First, thanks for the translation! But I encouter 
```
./bCNC
/home/tuxun/Bureau/bCNC/ProbePage.py:1201: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get().encode("utf8")))
Traceback (most recent call last):
  File "./bCNC.py", line 2316, in <module>
    application = Application(tk)
  File "./bCNC.py", line 191, in __init__
    page = cls(self.ribbon, self)
  File "/home/tuxun/Bureau/bCNC/CNCRibbon.py", line 106, in __init__
    self.register()
  File "/home/tuxun/Bureau/bCNC/ProbePage.py", line 1286, in register
    (ProbeCommonFrame, ProbeFrame, AutolevelFrame, ToolFrame))
  File "/home/tuxun/Bureau/bCNC/CNCRibbon.py", line 125, in _register
    w = f(self.master._pageFrame, self.app)
  File "/home/tuxun/Bureau/bCNC/ProbePage.py", line 1120, in __init__
    self.loadConfig()
  File "/home/tuxun/Bureau/bCNC/ProbePage.py", line 1153, in loadConfig
    self.set()
  File "/home/tuxun/Bureau/bCNC/ProbePage.py", line 1158, in set
    self.waitChange()
  File "/home/tuxun/Bureau/bCNC/ProbePage.py", line 1201, in waitChange
    CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get().encode("utf8")))
ValueError: 'AVANT & APR\xc3\x88S sondage' is not in list
```
when I use It. Maybe a little fix to do here?

tuxun, 2016-03-03:
Hi, I think every user with LC_ALL=fr_FR can't run bCNC because there is a "È" (forbidden in unicode) instead of "E" in the french translation. I "pull requested" the correction of the .po but forgot the updated .mo
(see post above, "APRÈS" instead of "APRES" give:
 ```
    CNC.toolWaitAfterProbe = int(TOOL_WAIT.index(self.toolWait.get().encode("utf8")))
ValueError: 'AVANT & APR\xc3\x88S sondage' is not in list
```
)

Also, in french, capitalized characters doesn't take accentuation :)

ThierryM, 2016-03-03:
Hi,
@tuxun
I don't agree with you : in french, the capitalized characters take accentuation. See here : https://fr.wikipedia.org/wiki/Usage_des_majuscules_en_fran%C3%A7ais
> "En français, « l'accent a pleine valeur orthographique ». L'Académie française recommande donc l'usage d'accent ou tréma sur une majuscule, tout comme l'utilisation de la cédille et de la ligature. Ainsi les publications de qualité écrivent-elles les majuscules (tout comme les capitales) avec les accents et autres diacritiques, au même titre que les minuscules.

I don't have the problem you report about the È : bCNC works with LC_ALL=fr_FR (it's working under Ubuntu 14.04 / Debian Wheezy).
Besides È exists in Unicode : È -> 00c8. The interest of Unicode is precisely to have the capitalized characters.
So I think that your enhancement is rather a regression.
Perhaps you have a problem with your configuration ?
Regards

CarlosGS, 2016-03-03:
I was about to write the same thing: I'm not able to reproduce this problem
The spanish translation uses accents as well, and they do not seem to be an issue

**Would you mind giving us some more insight regarding your setup? You can copy the system info that appears in "File->Bugs"**

Also, in #253 the .mo would need to be updated as well for the change to have any effect (see [how to use poedit](https://github.com/vlachoudis/bCNC/wiki/How-to-contribute-a-translation) )
In any case, thanks @tuxun for taking the time to help ^_^

vlachoudis, 2016-03-03:
OK I reverted the pull request. But we have to understand why it failed to recognise the character.

tuxun, 2016-03-03:
Hi, and thanks for your answers
It helped me to think about it in different way.
I also "try and test" in poedit.

When I paste an "È" char in another string, and remove the "È" in the primary faulty message string,bnc start.
When I paste this char in the "before and after probing" message string, It "bug" again on it.

So... it finally look like this string in particular isn't protected the same way than other (or there is another mystery here)

If you're agree with it, I would dig a little more (e.g. read code in probingpage.py), to check that point and propose a better patch, there is my firsts one (as you maybe already guessed)

tuxun, 2016-03-07:
@vlachoudis removing encode("utf8") from line 1201 in probepage.py allowed to launch bCNC without problem. But obviously, it introduces other bug...

[image]

Maybe a "setint()" problem with the index string of this message string?

vlachoudis, 2016-03-07:
Could you replace the encode() to decode("utf8") ,and check

ThierryM, 2016-03-07:
Hi,
The problem of @tuxun doesn't occur on my laptop (Ubuntu 14.04) : but I'm not connected to my machine (maybe I could test this wednesday).
[image]
So I can't help you ...
Regards,

Thierry

vlachoudis, 2016-03-07:
I was able to reproduce it, but it is very weird.
If I set the LANGUAGE=fr and have in the bCNC configuration to get the language from the system it gives the traceback.
If I don't set at all the LANGUAGE and change to French from bCNC settings it doesn't give the traceback.

tuxun, 2016-03-07:
@vlachoudis  I now think that it works if we use encode("utf8") in loadconfig() and decode("utf8") in saveconfig() instead of in waitchange() function

vlachoudis, 2016-03-08:
@tuxun I think I've found the problem. In the Utils.py: __bulitins._ I was setting it to ugettext, returning a unicode object, while when loaded from the ini file I was setting it to gettext. By changing it to gettext the first reference it seems it works for me. Could you check it?

tuxun, 2016-03-09:
@vlachoudis everythings looks good with the last commit! Thanks to all!

ThierryM, 2016-05-28:
Hi,
See the screen with words (in red circles) not translated in french. It's quite strange because "Sonde" is correct (Probe in english).
_Now, an error existes in french translation ("aud" in place of Baud) : it will be corrected in the next version._
[image]
Regards,

Thierry

closed 2017-05-11 by vlachoudis