vlachoudis/bCNC #154: Stop more easily the probe scan

ThierryM, 2015-11-18:
Hi,
Thanks for your work and sharing.
I think that it will be easier to stop the probe scan without having to go on the control tab and the stop button.
Regards,

Thierry

vlachoudis, 2015-11-18:
You can add the "Stop" buttons in the "Probe" page, by changing the .bCNC config
look the bCNC.ini and search for probe.ribbon. Copy this line in your .bCNC config file and add there the Run keyword, and you can have the "Run/Pause/Stop" controls in the Probe page.

ThierryM, 2015-11-18:
Thanks a lot for your quick help.
Your solution works fine. So the GUI is highly and easily flexible : an other very good point of bCNC.
Regards

vlachoudis, 2015-11-18:
But It needs to be documented on the wiki

closed 2015-11-18 by vlachoudis

Effer, 2015-11-18:
What about a contexts menu like the one in the status button?
It would be nice to customize the ribbons as easy as right click and check what you need. Maybe after restart..

ThierryM, 2015-11-18:
I have added this "trick" in the wiki "Configuration part".