vlachoudis/bCNC #155: Arduino/serial based pendant

Nate-Bro, 2015-11-18:
I'm not sure if it's been discussed before, but what all would be involved in connecting to a second serial device, let's say a second arduino, that has the usual pendant buttons, and electrical E-Stop buttons.

Right now I have a pendent that has the E-Stop cut off, Abort, resume, hold buttons. That connect to the arduino  cnc shield. But it would be nice to have a second arduino with control buttons and buttons that send commands to trigger custom actions.  I could use the same arduino, but where is the fun in that.  

That way I can use the hardwired pendent when I'm next to the machine, then use my phone/web pendent when I'm across the shop or in another room.

vlachoudis, 2015-11-18:
A simple solution to that would be to make an external program, that communicates with the second arduino and sends the commands to bCNC simulating a web pendant.

Nate-Bro, 2015-11-18:
So an option would be say have an arduino with a lan/wifi shield communicating with the Web pendent.  

That's a good solution, and something I could work on while working on the pendent. 

Thanks

jdiderik, 2015-11-18:
I've been thinking about the same thing, maybe on the basis of an ESP8266
The web pendant is on hold for now, I really need to actually do stuff on the machine.
But as soon as I can find some time and we can get websockets and with that the response of the web-pendant usable (for me), jogging and steering my cnc  from a hardware pendant can become reality :)

vlachoudis, 2015-11-18:
@Nate-Bro a lan shield could do the trick. I was proposing a second Arduino with usb connection to the same computer running a dedicated python script to communicate with bCNCC.

dshookowsky, 2015-11-29:
I'd like to have something physically wired to the raspberry PI when I'm using bCNC for eStop,etc.  It will be an almost headless install.  I don't trust the wifi in my house to work reliably enough that a phone/web-based eStop would work.  I'd prefer to have something listening for Arduino Serial communication on the PI side and then sending web-based pendant commands to localhost.  I happen to have a spare Arduino and PI2 handy and I'll see what I can get wired up.

chamnit, 2015-12-11:
I just caught wind of a new Grbl GUI called [cnc.js](https://github.com/cheton/cnc.js). It's basically Chilipeppr that works offline. I haven't tried it yet, but at first glance, it looks like it's been done well. I wonder if this could be plugged into bCNC's powerful backend as its web pendant somehow with minimal effort. Anyone want to give this a stab?

Nate-Bro, 2015-12-11:
I'm going to give that a look

https://github.com/vlachoudis/bCNC/issues/155#issuecomment-163968524

vlachoudis, 2016-01-21:
Look at this video [https://www.youtube.com/watch?v=dYixko-8JMY] he is using an XBox controller as a pendant to bCNC

Effer, 2016-01-23:
Back to @Nate-Bro original question:
My proposal is to implement a background thread in bCNC that keep pooling a second USB Arduino running firmata firmware. Firmata is a standard protocol adopted by many to exchange in a fast and optimized way I/O over a serial protocol. The library is already present in the Arduino IDE (and supported) and allow to configure I/O (input, ouput, pwm, analog in and even servo or stepper) and write/read them.
Binding Inputs to bCNC command (run,stop,jog etc) will help to create custom hardware pendant that users can personalize. 

A future development could be even map some special M code to digital output (M100 --> Digital output 5) and in this way realize commands synced with gcode to expand grbl (and implements some interesting automation).

Probably it needs to keep in mind RPi users that would like to use the on board GPIO, so an hardware abstraction layer would be a nice addition to cover more hardware later.

Python library for firmata exist, even some simplified version single file based, so an implementation seems not so hard.

vlachoudis, 2016-01-23:
@Effer this could be one option, however it will limit its use. If we create a TCP/IP socket interface (maybe compatible with web sockets) as the present web server, without the need to drop/restore the connection on every request, it will be more responsive than the present web interface and easier to integrate with any external program. 
Making a TCP/IP server that accepts requests from the net is quite trivial, I have not investigated what the websockets require.

Effer, 2016-01-23:
@vlachoudis TCP/IP interface has advantages over serial approach but IMO it have almost 2 big negative facts:
- the price (even if chines Ethernet shield are very cheap nowadays and ESP8266 is an interesting alternative)
- the develop of custom firmware (running on the hardware pendant device), while firmata is ready made.

It's a true fact that a fast TCP/IP socket will offer a responsive interface over any smartphone. But when it comes to interface with something physical (buttons, joystick, tool change) the problems rise.

Probably here are discussed two different needs, one is interface with other software, the other is interface with something physical in a economical way.

vlachoudis, 2016-01-23:
I didnt even thought of connecting with the pendant over Ethernet. Rather to have an easy and fast way external processes to communicate with bcnc. My approach would be to have a prednant with Arduino with our without firmatta, Bluetooth etc connecting to a local process that sends the requests to bcnc via TCP/IP

Effer, 2016-01-26:
@vlachoudis

If I well understand a new thread will take care of websocket (or TCP/IP) maybe substituting the existing HTTPServer. It will expose a software interface where write commands that will be put in the app.queue
Any new "pendant plugin" will retrive inputs from the field and then will push command to the socket. It will also need to expose the same socket interface to retrive command from bCNC. (or keep pooling for them).

But is a bad thing append directly to that app.queue from the pendant plugin? Where do the user will specify what to bind to the hardware pendat? In the pendant plugin or in a bigger shortcut list?

I am still very uncertain about the best approach to solve this requests, really very interested to see how will be solved.

stanoba, 2016-01-30:
I have chosen different approach. Arduino MEGA and 2.8" touch LCD as pendant. Now I'm working on arduino sketch which emulates USB keyboard. (arrow keys at the present moment)

@vlachoudis - Is it possible to implement more keyboard shortcuts?

[image]
[image]

vlachoudis, 2016-01-30:
Very interesting @stanoba. It looks a bit like the web pendant. As the program is right now you can bind any command on the F# keys either pressed alone or with Shift / Ctrl. Look on the "Tools" tab "Shortcuts"

Nate-Bro, 2016-02-03:
Can it be implemented with a no lcd option. Like if we just wanted to use a keyboard and buttons.

https://github.com/vlachoudis/bCNC/issues/155#issuecomment-177025148

stanoba, 2016-02-03:
@vlachoudis - Thanks for info.

@Nate-Bro - Yes, you can implement cheap Arduino Pro Micro ATmega32U4 which has native keyboard and mouse support (HID): https://www.arduino.cc/en/Reference/MouseKeyboard

I can also imagine combination of buttons and rotary encoder.

auhopu, 2016-02-04:
I was wondering what kind of assignments are possible with shortcuts and if there is any formatting consideration.

For instance can one assign
- g-code directly?
- one of the 6 user defined buttons (g-code indirectly)?
- GRBL commands?

Effer, 2016-02-04:
I believe it accepts almost all commands here : https://github.com/vlachoudis/bCNC/wiki/CommandLine
Tested with F10 --> RUN, F11 --> STOP
never try with gcode

Nate-Bro, 2016-02-04:
I think having an arcade style joystick would make things simpler for me. But it seems to have an issue with multi direction. When u use the keyboard it wants to run eather the y or x, if I press both it just moves in one direction, but not together.

I used one controler program that gave my machine a nice smooth x and y jogging.

https://github.com/vlachoudis/bCNC/issues/155#issuecomment-179511024

auhopu, 2016-02-04:
@Nate-Bro
This is a funny coincidence... I had the exact opposite problem! :)

Due to the nature of the PS2's analog joysticks it was impossible to get a clean X or Y movement. I noticed in jtest that whenever I moved the joystick in one axis there was always a slight move in the perpendicular axis. I ended up filtering it out by setting an 80% threshold in the pygame script, because I personally consider this noise. Since I map the jogging moves to the Up/Down/Left/Right keys, I prefer to be sure that I only move Up/Down/Left/Right with each press.

I guess it drills down to whether you prefer fast diagonal moves or precise moves per axis. On a joystick with both analog axes and cross-layout buttons, you could probably program both.

dshookowsky, 2016-02-04:
I know from my MAME arcade cabinet days that Happ makes switchable 4/8 way
joysticks.  There's a piece that physically limits travel to 4 directions.
When removed, the full 8-way control is allowed.
https://na.suzohapp.com/products/joysticks/50-6084-000

vlachoudis, 2016-02-04:
@Effer everything is accepted (like in the web pendant) commands or gcode or even the "SENDHEX" command to send character encoded in hex to grbl.

auhopu, 2016-02-04:
Quick question: will a second PAUSE from command line trigger a resume? I.e. will the first send GRBL a "!" and the second a "~"?

vlachoudis, 2016-02-04:
No. Pause sends !, resume sends ~

auhopu, 2016-02-04:
How do you send resume from the command line? I do not see RESUME on the list. Can you send a plain "~"?

PS. Sorry for bugging you with this. I am currently setting up a CNC, but the GRBL controller is on its way, so I am just trying to sort out things in my head without having an Arduino yet.

vlachoudis, 2016-02-05:
@auhopu The wiki is outdated :)
bCNC accepts the following:
PAUSE:   equivalent to the pause button toggle between feedhold / resume
FEEDHOLD: sends an !
RESUME: sends a ~
Also you can send from the command line the ~ and the !

Effer, 2016-02-29:
@vlachoudis after some thinking, I'm totally agree in the fact that bCNC should have a public interface to play with. Any future plan about it?

vlachoudis, 2016-03-05:
A nice and easy alternative of a hardware pendant is a wireless numeric keypad as it was pointed in #244.

twforeman, 2016-03-08:
More info on using a wireless num pad here: http://www.shapeoko.com/forum/viewtopic.php?f=5&t=7507&start=50#p61047

I've been doing this for a little while and it works great.

Effer, 2016-03-08:
Nice job @twforeman and Interesting build log.

WillAdams, 2016-04-06:
Any love for the Contour Design Shuttle Express?

It's actually rebadged and sold as a pendant w/ some commercial CNC setups.

Buttons needed are XYZA axis selection, speed toggle (cycling through speed options) and support for the inner and outer jog wheel.

vlachoudis, 2016-04-07:
I never tried. Right now with my machine setup I don't have the need of any pendant, but if someone is willing to work on, I can help.

njkeng, 2016-05-29:
I own a Contour Shuttle Express and I found it very useful when I was using ChiliPeppr.  There's a utility that was written to make the comms happen between the Shuttle and the Serial Port JSON Server that is used by ChiliPeppr.
I missed using the controller when I made the switch to bCNC.  Since I already had the controller, I thought I'd try to modify the Shuttle utility to talk to bCNC in addition to ChiliPeppr/SPJS. 
Long story short, I have the Shuttle working nicely with bCNC, thanks in very large part to the original ShuttleCP project: https://github.com/lordmundi/shuttleCP
I created a new branch of the shuttleCP project in git to develop the bCNC interface and I have now put in a pull request to see if it can be officially incorporated. I'll keep you updated.

vlachoudis, 2016-05-29:
Fantastic @njkeng

lordmundi, 2016-06-10:
The updates from @njkeng are now in the master branch of shuttleCP so feel free to try it out from the latest version.

comps, 2018-07-27:
(Wanted to post this as a new issue, but then found other hardware pendant ones being closed in favor of this one, so just adding here.)

The most universal solution for a "true" hardware pendant seems to be the joystick interface in various OSes, not a custom serial/network/json/arduino/etc. protocol.

As a program running on an OS (at least Linux and Windows), reading a joystick, you can have 1-dimensional analog wheels, 2-dimensional analog stick movements, many buttons, etc., all via the same interface. Even the calibration is done for you by the OS driver (if supported).

Using a joystick API also gives great hardware freedom to users, as potentially any PC gaming accessory uses the same API or has other programs/drivers that do the translation (ie. for PS4 controller on Windows).
In the extreme case, you can wire a 3-pin rotary encoder to GPIO and use a device-tree driver to simulate a 1-axis input device, completely transparently to the program reading the input (bCNC).

Just imagine mapping a 1-axis throttle control on a joystick to ie. feed override.

All this would probably need to use Grbl 1.1 and its "joystick implementation" recommendations, https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging#joystick-implementation , instead of moving fixed distances with multipliers.

I realize this is probably not a simple fix, but it would allow for real low-ish latency jogging experience using widely available consumer hardware, without relying on specific OS or additional control hardware (RPi).

MildaGenius, 2018-08-21:
Hi,
Iam testing first version of Arduino pendant connected to bCNC. It uses serial connection to Arduino and web pendant from bCNC (bridge). I have opened connection to web pendant, where I send commands and from the same connection refresh CNC positions. Web pendant and bridge are on the same computer.

Main problem is lag between button press and response in bCNC. Its about 2s :(.
Second issue is, that bCNC dont react on spindle on/off (I see success request, but nothing happend).

[image]

https://youtu.be/qBP-9yQEiUU

aggiedan977, 2019-04-12:
I used this method to connect a USB gaming controller to my pi and jog my bcnc: https://openbuilds.com/projectresources/raspberry-pi-xbox-controller-bcnc-and-you.259/

agent-r, 2021-02-09:
Here some informations on a working (but still work in progress) bCNC pendant project.

[https://github.com/agent-r/bCNC-PENDANT](https://github.com/agent-r/bCNC-PENDANT)

Harvie, 2021-02-25:
I would now reccomend everyone to use grblHAL with some pendant plugin to have pendant connected directly to machine without having to go through PC https://github.com/terjeio/grblHAL/

lordmundi, 2021-02-25:
Wow... that's a neat development!  I wasn't aware GRBL was going that direction.  For folks looking through this thread, here is the article on the grbl website describing the project: https://www.grbl.org/what-is-grblhal

I hope that this will merge in or work with the work Bart has done on GRBL ESP32 and his awesome Universal modular CNC controller: https://www.tindie.com/products/33366583/6-pack-universal-cnc-controller/

Harvie, 2021-02-25:
> I hope that this will merge in or work with the work Bart has done on GRBL ESP32

I hope also, but i am not sure if this is going to be easy, because Grbl_ESP32 fork is heavily optimized for use on ESP32 and doesn't really care about other micros, while grblHAL fork is more generic and tries to support many different 32b uCs. So there might be some conflicts of interrest until everything gets sorted out properly. Hopefuly they will be able to unite their efforts...