vlachoudis/bCNC #157: DXF properties

dodolbysuround, 2015-11-19:
Hi,
I would like to know how to configure my DFX files so that they can be imported into bCNC.
I manage to open the ones I draw with inskape but enconter problems trying to open the ones I generate with solidworks.
Thanks for your help

vlachoudis, 2015-11-19:
Hi @dodolbysuround could you send me one of the dxfs that are not able to be imported to check.

dodolbysuround, 2015-11-19:
hi, thank you for this quick answer.
you'll find in attachement a DXF file directly saved from solidworks.
I also join a picture of the differents available option in solidworks
the "version" scroll list include:
-R12
-R13
-R14
-2000-2002
-2004-2006
-2007-2009
-2010
Looking forward to read from you

Date: Thu, 19 Nov 2015 03:04:42 -0800

vlachoudis, 2015-11-19:
Sorry but I cannot find the attachment

vlachoudis, 2015-11-19:
it loads the dxf normally with no error. Is it correct?

[image]

dodolbysuround, 2015-11-19:
It seems to be correct indeed on your picture but when I do open it, nothing display, execpt from 'c:\user\loic\descktop\Pièce1.DXF' loaded in the left bottom corner

Date: Thu, 19 Nov 2015 09:23:26 -0800

vlachoudis, 2015-11-19:
Maybe is the e with accent in the filename

dodolbysuround, 2015-11-19:
I tryed without the accent ant that's the same.
What's amazing is the that in your screenshot, only the name of the file appears. In my case, the whole path displays.

dodolbysuround, 2015-11-19:
Ok, problem solved:
the extension must be written lower case and by default, solidworks write it upper case.
then Pièce1.DFX doesn't works but Pièce1.dfx is ok (-;

vlachoudis, 2015-11-20:
Good to know. I will modify the code to accept upper and lower case extensions

vlachoudis, 2015-11-20:
Ok now it is fixed. Accepting both lower and upper case extensions.

closed 2015-11-20 by vlachoudis