vlachoudis/bCNC #162: More buttons

onekk, 2015-11-28:
Hi @vlachoudis I'm using bCNC again after a stop due to a remanufacturing of the controller electronics, I have to flatten my bed, so i have to define some predefined positions on the workspace to test the bed heihgt with a dial gauge, and then put some shimming under the table.

I have seen that in the Control interface there is some space for more user buttons, so adding some more buttons can help me a lot, in the code I have 

    class UserGroup(CNCRibbon.ButtonGroup):
    	def __init__(self, master, app):
		CNCRibbon.ButtonGroup.__init__(self, master, "User", app)
		self.grid3rows()

		n = Utils.getInt("Buttons","n",6)
		for i in range(1,n):
			b = Utils.UserButton(self.frame, self.app, i,
					anchor=W,
					background=Ribbon._BACKGROUND)
			col,row = divmod(i-1,3)
			b.grid(row=row, column=col, sticky=NSEW)
			self.addWidget(b)
  

if i change the 	line  n = Utils.getInt("Buttons","n",6) modifying the 6 i think it will do thw job, but the interface is not large enough to contain the added buttons, where are the definition for the dimension of the user "panel"?

TIA and Regards

Carlo D.

vlachoudis, 2015-11-28:
HI @onekk 
you should change your .bCNC file instead in the section [Buttons] add a field
n=### the number of buttons you want

closed 2015-12-02 by vlachoudis

Fosforitofercho, 2017-02-19:
Thank you so much, pretty easy , we need a manual for the config file, is incredibly powerfull