vlachoudis/bCNC #163: Autolevel Issue

atrueresistance, 2015-11-28:
Hello, I just installed limit switches on my SO3. I was able to complete a tool change just fine. When trying to autolevel I get this error. I'm not sure if you need the files... Guessing it is user error
[image]

---------------------------
State: error: Invalid gcode ID:24
---------------------------
Last error: g43.1z-6

Two G-code commands that both require the use of the XYZ axis words were detected in the block.
---------------------------
OK   
---------------------------

vlachoudis, 2015-11-29:
Hi @atrueresistance if you click on the red "Status:" it will tell you which line caused the problem and also the explanation of it he ID:24

vlachoudis, 2015-11-29:
@atrueresistance I've introduced a bug when I added the to accept unicode file names. The unicode strings in the gcode were ignored and concatenated with the next one. Therefore the two commands were together. Now it is corrected.

atrueresistance, 2015-11-30:
Thank you! I can confirm autolevel working. Can you go into detail on the tool change options. I have setup a tool change area and a probing area. I chose the setting to pause before and after. When I hit the macro the machine moves to those areas, but does not actually pause. 

Currently if I load a project, do not autolevel, and hit go. The machine will move to home area, move to probe, probe, and then go directly to cutting without pausing. To me that isn't very safe as I have to remove the alligator clip and turn the router on while in motion. The first time this caught me really off guard as I wasn't expecting it to continue. 

Autoleveled, the machine will go straight to cutting.

atrueresistance, 2015-12-01:
Also grbl throws an error when encountering a m6 command after an autolevel. Under Probe -> Tool I have the policy set to Manual Tool Change (TLO) and pause before/after.

chamnit, 2015-12-01:
@vlachoudis : M6 is not a supported g-code by Grbl. So it will throw an error. bCNC will need to remove it from the stream unfortunately. I'm not quite sure how to otherwise to do it, at least within Grbl itself without mucking up things.

atrueresistance, 2015-12-01:
I thought that bCNC intercepts the M6 before sending if you specify manual tool change. Ie running the macro. If there were any changes today to master I didn't download. The version I'm on is two days old, right after the Unicode fix.

vlachoudis, 2015-12-01:
If you select the "Manual Tool Change" it should substitute the m6 with the appropriate sequence. If not there, then something broke there. I will try to find it.

vlachoudis, 2015-12-03:
@atrueresistance I've made a few tests on my machine and a clean installation and I didn't find any problem. Could you send me your .bCNC file to see if some particular settings are causing the problem.

atrueresistance, 2015-12-10:
Yes I can. When I get off work I can upload.

atrueresistance, 2015-12-12:
[bCNC.txt](https://github.com/vlachoudis/bCNC/files/59998/bCNC.txt)

vlachoudis, 2015-12-12:
What you've send me is the default bCNC.ini and not the .bCNC with your preferences.

atrueresistance, 2015-12-17:
[image]

Otherwise I'm not sure what file to upload

Effer, 2015-12-17:
Hi @atrueresistance I don't have an english copy of windows but the fast way to get there is to pop up command windows with the Windows Key + R
Then write this command %UserProfile% and hit enter.
It'll will show a folder where you can find the .bCNC file

[image]

Or if you want to go with explorer should be in
C:\Users\YourUser\

mano1979, 2015-12-19:
I did fix the "not pausing" problem for grbl.

**In the bCNC folder, open the CNC.py file with a text editor.
goto line 1400 and change:**

if CNC.comment:
lines.append("%%pause Tool change T%02d (%s)"%(self.tool,CNC.comment))
else:
lines.append("%%pause Tool change T%02d"%(self.tool))

**to this:**

if CNC.comment:
lines.append("m0")
else:
lines.append("m0")

** Now pause is working. To "resume" rightclick on the info bar (run, idle, error, alarm) and click on Resume.**

vlachoudis, 2015-12-20:
Can you try instead of sending chaning the CNC.py to change the Sender.py line 633 and replace the "!" with a "m0\n"

mano1979, 2015-12-20:
Indeed! That works also. Not noticing any difference though, but i'm sure this is better as what i did.
Also i found that the grbl firmware can be used unchanged. No lines of code have to be added.

vlachoudis, 2015-12-21:
This way it displays also a message on the status to change the tool, otherwise it will simply mention paused.

mano1979, 2015-12-21:
I see. I am not sure if it is working though. Atleast the message is not displayed while in pause. It might have flashed by though...

closed 2016-02-12 by vlachoudis