vlachoudis/bCNC #164: Please ZERO any location of the probe before starting a run

atrueresistance, 2015-11-28:
Hello, I don't seem to be getting this right. I ran the probe function, but still can't hit Start without this error. 
[image]

[image]

vlachoudis, 2015-11-29:
After the autolevel probing cycle you should zero the surface (z-coordinate) at any point you like and then click the "Zero" on the autolevel probe. Then the autoleveling information will be relative to this point.
It is a bit of redundant information as a safe measure, normally it is needed in the case that you load the probe from a file.

closed 2015-11-30 by vlachoudis

atrueresistance, 2015-11-30:
Thanks! Added a note in the wiki for new people, as it wasn't clear at first to me.

vlachoudis, 2015-11-30:
I have to make it more smart in the future. That if it was zeroed before the probing cycle to remember the information.