vlachoudis/bCNC #165: Edit the Header and footer and feed for laser cutting

asm7100, 2015-12-03:
Hi how to edit the header and footer?
Is it possible to make a tool, named laser cutter, set the hight of it (z axel) and the feet af it and also the M3 S255 to full power the laser. 
I have a Spindel and af laser on my own build machin, and control the laser with M3 S255 (Full power) of the laser, and stille use the M3 / M4 Sxxxxx to the spindel

Effer, 2015-12-04:
Hi @asm7100 .
You can easy change the header and the footer using the tool tab and the settings button:
[image]
The header and footer are added when you create a new file.

Do you need to change the header and footer when using the spindle or the laser? 
If so maybe we could add multiple header and footer and use them in relation to the "LaserCutter" option?

asm7100, 2015-12-04:
It could be nice if some how could use the M3 / M5 to turn laser on / off, lets say you set the M5 (Laser off) before you quick move the laser, and when its slow down, set M3 to turn on the laser.

vlachoudis, 2015-12-04:
If you tick the "Lasercutter" (second line from the top) and you load a dxf file it will turn off before the g0 codes and turn on before the cutting.

asm7100, 2015-12-04:
Yes i know, but you need to set the s parameter also. Just like
M3 S255, (S255 is high power for my laser)

vlachoudis, 2015-12-04:
Ok I understand, now the code is issuing only a M3/M5. I will add also the "S" statement after each M3 for the lasercutter then.

asm7100, 2015-12-04:
Yes thanks and may a Z Code to set the focus hight of the laser. But how about make it Make a "M3 Header for laser."

vlachoudis, 2015-12-04:
I was wondering, if you set once the Sxxx in the header it will not be remembered for all subsequent M3 commands?

asm7100, 2015-12-04:
That i dont know :-) But i cut be nice, is you can set it on the the laser tool, so that you can have more than one laser tool, on for full power and maby one for 50% power, do you get the idea.

Effer, 2015-12-04:
Maybe we can take the spindle/laser set from the material database instead? When lasercutter is set append that to M3 commands?

asm7100, 2015-12-04:
I need a way to tell my laser go to Z before cutting. (To focus the laser best)

Effer, 2015-12-04:
What about stock material safe Z?

asm7100, 2015-12-04:
Will the other thing not give more freedom?

Effer, 2015-12-04:
I think that all your needs can be menage using most of the already present settings, adding just a couple of fix. I'm just trying to reuse most of the base code, in this way an update to the cnc side will lead benefits (or bug fix) to all users (laser and not). Also to avoid duplicated settings.

My propose is to add a spindle settings to the material that can be used for both laser and cnc. Use  the "already there" safe z in the stock material as focus height for laser (and keep as safe height for cnc).
In this way if you need a full power laser you can create a custom material (with specific safe height and spindle power) and when "soft cutting" simply use another material.
Also in the stock material the "Surface Z" could be use as additional  offset for laser focus if you are working taller material.

asm7100, 2015-12-04:
Its more than you can use less power if you are using your laser to ingrave (Differences shades) "From black to gray)

Effer, 2015-12-09:
@vlachoudis how do you prefer to proceed?
I think laser will be mine next upgrade (don't know exactly when, but it's on my wishlist). It would be nice to have all software ready in advance of months :)

vlachoudis, 2015-12-09:
@Effer I think your proposal to use the surface Z of the material for the focusing of the laser is nice and easy to be done. It would require the CNC.zentry routine method to be modified and add a g0 with zsurface.

atrueresistance, 2015-12-10:
I have used the laser part of bCNC a decent amount. I swap between a dewalt
611 and laser mostly on a project needing cut out, then personalized. Some
differences I've noticed, when not in laser mode hitting go moves to tool
change position and then moves to WCS Z0. When in laser mode I typically
focus  and set WCS Z0. If Z is changed before  pressing go, the laser Z
does not move to WCS Z0. How hard would it be to attach an offset rod that
conducts to probe Z on curved surface?

asm7100, 2016-01-03:
I you can set a Z value for the laser tool (Focus point) and then take the material hight and add it to the Z :-) A footer / header for laser nice and a way to edit the footer and header (I cant edit my header and footer :-(

closed 2016-03-05 by vlachoudis