vlachoudis/bCNC #168: Setting Zero Point with camera

danst0, 2015-12-07:
Hi,

Maybe I just oversaw, but is there a way to use bCNC together with a camera attached to the spindle to set the work piece location/zero point?

Daniel

vlachoudis, 2015-12-09:
Hi @danst0. No there is no such option.

danst0, 2015-12-09:
Too bad. However also not a killer feature. 

How far is the camera integration into pendant? I did not find any howto on that. 

Daniel

chamnit, 2015-12-09:
@danst0 : I've also seen people using webcams to do XY zeroing as well, but I don't think you really need anything special to do this. You simply need window with the camera in view and then zero the axes out with bCNC. Any program can do this task for you.

danst0, 2015-12-09:
I am not so familiar with the Linux program landscape in this regard. Anything you could recommend, that is not too much of a hack?

Daniel

Effer, 2015-12-09:
When I first read the post I though of something different:
A macro will scan an area looking for the qr code. Decoding qr code size with specific pixel/size camera ratio and some math could detect the XY offset for zero, maybe also Z when camera is calibrated http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

Using opencv library (already choose for webcam implementation ) and a qr code of fixed dimension could make the trick.

Really poor probability of any utility.

closed 2016-01-06 by vlachoudis