vlachoudis/bCNC #171: Autolevel probes do not fill workarea

puppyscrubber, 2015-12-13:
Hi, I couldn't find a forum to post to first and I hope this is not a user issue!
When doing autolevel, the probes happen throughout the specified areas but the green probe result distances do not show for the last part of the autolevel process. The values are shown in the gcode in the terminal but not on the 3D view and when saved to probe file, these last probes in probe file are all zero's. 

Is this a bCNC or user issue and is it possible to, or how do I fix this? (I have both specified and loaded margins from input file and cleared and zero'd and this happens for small and large number of probe points.)

Many thanks for great software! :-) 
See image.
[image]

vlachoudis, 2015-12-14:
It is strange. Based on what you say it seems that the it failed to recognize the PRB string returned from grbl. Is it something reproducible, if yes can you send us the steps to reproduce it?

puppyscrubber, 2015-12-21:
Thanks for feedback!
It seems like this is some sort of communication error/issue because Terminal log shows something going wrong near end of transactions. See image below for notes.
Is this not linked to proposed issue #175 ?
This issue did not happen previously with my setup (It might not happen with previous bCNC  builds or older grblForCyclone builds)
Issue on both WinXP and Win7. (bCNC 0.7.2, I think also with 0.6.6 but will have to confirm still)

***The "progress bar/time remaining" goes to 100% where the green numbers stop and then just says "Run Ended" while the CNC machine completes the rest of the probes without returning any green probe distances.*** 

[image]

tiltit, 2015-12-22:
Hi, exact same problem here, running on Linux Mint. I just rolled back to bCNC 0.5 and it does not have the same issue. 
I love bCNC by the way, it's the reason why I will probably never upgrade to Linux CNC. Thanks.

puppyscrubber, 2015-12-25:
Hi Vasilis,
More info on this as requested is below:
No auto-leveling issues/all OK up until 23rd Nov b40f640 commit, where Bcnc starts to complain about " invalid Gcode ID 24" and never actually starts the first probe so I cant tell if 72% issue starts here or not.
Then on the Nov 29 commit 6aa4ac; the "invalid GCODE ID 24" error stops and at this point, the 72% autolevelling issue starts.

I hope this helps and let me know if I can do anything else to find and fix this. I'm not new to programming but *very* new to python so I would need to get over python learning curve before I can directly help.

Many Thanks!

vlachoudis, 2015-12-28:
Thanks for the investigation. I will give a look once I have access on my machine to check.

se7en9057, 2015-12-28:
i can confirm the same issue for me running on Lubuntu 15.10

gcobos, 2016-01-10:
Hello, 

The problem arises when bCNC finishes sending all the PRB commands to Grbl, and sets the flag self.running to False. Grbl is still executing the last probes, and sending the responses, but bCNC is ignoring them. As a quick fix, just getting rid of the condition "if self.running:" in the file Sender.py solves the problem.

diff --git a/Sender.py b/Sender.py
index 3e1d35a..6913404 100644
--- a/Sender.py
+++ b/Sender.py
@@ -725,7 +725,7 @@ class Sender:
                                                                CNC.vars["prbx"] = float(pat.group(2))
                                                                CNC.vars["prby"] = float(pat.group(3))
                                                                CNC.vars["prbz"] = float(pat.group(4))
-                                                               if self.running:
+                                                               if True: #self.running:
                                                                        self.gcode.probe.add(
                                                                                 CNC.vars["prbx"]
                                                                                +CNC.vars["wx"]

vlachoudis, 2016-01-11:
Thanks @gcobos. This is how it was in the past, but I've added the condition as a safe check to avoid probing points included in the autolevel after the probe. I think I can remove from the time being until I get access to my machine to check and find a more robust solution

vlachoudis, 2016-01-11:
Ok. I removed it, restoring it to the past state.

vlachoudis, 2016-01-17:
This is duplicate of #175 and I believe now it is fixed (at least temporary) until a more robust solution is found to detect the end of a run

closed 2016-01-17 by vlachoudis