vlachoudis/bCNC #174: Autolevel disables toolchange (M6)

mano1979, 2015-12-18:
I have tried multiple times, but everytime when i have done an autolevelling scan and then after that try do start my gcode, the T0 M6 command is ignored. This means that there is no probing done before the mill job and the mill is miling air as a result.

Is this supposed to happen? Is there a clear tutorial/manual on how to set this up correctly?

PS: For tool change (manual wcs) to work with GRBL, in bCNC.py line 1400 and 1402, it should indeed use M0 instead of %pause.

vlachoudis, 2015-12-20:
You are right this is a bug that needs correction.
The autoleveling goes on a different branch that is ignoring the tool change expansion.
Since I don't have access on my machine for a few days to I will do the correction
and send you the code to check before pushing on git.

%pause is doing a bit more than M0. It waits until the grbl buffer is empty as well the internal queue is empty as well.

mano1979, 2015-12-20:
Ok that sounds great! I am mostly home and behind the computer, so i can try thing on the fly.

vlachoudis, 2015-12-20:
Could you please test the following https://dl.dropboxusercontent.com/u/85577225/CNC.py

mano1979, 2015-12-20:
Yes! This works! Toolchange command works after a autolevellng.
There is still one issue though. After probing and levelling the bit does not go deep enough.
It looks lke the milling depth is not taken into consideration.
My bit hangs just over the surface instead of 0.1mm below it.

mano1979, 2015-12-20:
Another problem i found yesterday while testing is that the M6 command is accepted the first time in the gcode file, but the second one is ignored and an error appears from GRBL that the M6 command is unknown. Wich in return fails to do a toolchange

vlachoudis, 2016-01-17:
With all the changes in the autolevel, now this one should be fixed. Can you confirm so I can close the bug?

mano1979, 2016-01-19:
I have downloaded the latest .zip from github but i can not load any gcode file.

The "open" button doesn't do anything when i click it. (The open button in the file browser)

Effer, 2016-01-19:
Hi @mano1979 
I've just tested opening some files and it's working for me. Do python report any error?

mano1979, 2016-01-19:
I don't know if python gave an error. After closing bCNC i got a window with an error though. I have send it to your website (automantic error report)

vlachoudis, 2016-01-19:
Today I didn't receive any error report. @mano1979 could you try it again starting from a terminal so python can output the error on the screen.

mano1979, 2016-01-19:
Here is the copied terminal output:

https://www.dropbox.com/s/ny03i417pjsrpbh/BCNC_error_log.txt?dl=0

vlachoudis, 2016-01-20:
Thanks @mano1979 I corrected it.

vlachoudis, 2016-01-23:
@mano1979  does  the M6 after the autolevel behave correctly now?

mano1979, 2016-01-23:
I'm sorry. I haven't had time to test it yet.
I can probably test it today. ;)

vlachoudis, 2016-01-27:
I've just tested it on my machine, and it works Ok.

closed 2016-01-27 by vlachoudis