vlachoudis/bCNC #175: Autolevel stop at 72% of workarea

TonTonqui, 2015-12-19:
Hello,
 I want to say thank you for this wonderful app. you do a great work. 
everything works great, except the "autolevel".
 it stop the log notes in 72% of the runs, but continuous measure it until the end. 
see attached screenshot. 

[image]
[image]

can you help me fix this issue? 
my setup is: win7 + the latest version of bcnc master + python 2.7.10. 
Thank you for help. sorry for my bad englais I use google translator.

vlachoudis, 2015-12-20:
The bar stops when all commands are flushed to grbl. However grbl has it own queue which will hold still a few commands. Maybe I should add a %pause at the end, so the run to wait until all commands are flushed and executed by grbl

TonTonqui, 2015-12-20:
hi,
thank you for your answer. That may be a good idea, this would empty the spool. Waiting your change, I increase the speed of probing to see.
Regards.

TonTonqui, 2015-12-20:
when I reduced the probe feed at 10 and bcnc log one more point only, and when i put the feed at 200 nothing change. can we increase the speed of grbl ?

vlachoudis, 2016-01-04:
With the new update, I believe it should solve the issue. Could you please check?

asawyer215, 2016-01-09:
Hi vlachoudis, I just pulled the master branch and I am still seeing this issue.

ThierryM, 2016-01-09:
Le 09/01/2016 20:49, asawyer215 a
      écrit :
    
    
      Hi vlachoudis, I just pulled the master branch and I am still
        seeing this issue.
    
    Hi,
    I have the same problem I think : I try to probe with 25 points but
    there's only 18 points saved (100% completed) : but the probing is
    going on to the 25th points. See the picture :
    
    
    Regards

ThierryM, 2016-01-09:
Hi, I have the same problem I think : I try to probe with 25 points but there's only 18 points saved (100% completed) : but the probing is going on to the last 25th point. See the picture : 
[image]

Regards

roboduck, 2016-01-15:
Hi, I noticed the same problem.
I did a local fix for myself by removing `self._gcount += 1` in Sender.py:651 (I am doing it at my own risk and I discourage doing it without understanding all consequences!).

There seems to be a discrepancy between how many commands are sent to grbl and the actual `gcount` number. During probing `gcount` reaches number of commands sent to grbl before they are actually sent, which indicates end of the run. Therefore any data for remaining probes received from grbl after this point are ignored (even when all the remaining probes are successfully visited).

vlachoudis, 2016-01-15:
The problem was coming from the condition to use the probe points during the run. By adding a condition WAIT at the end of the run I forgot to sync the number of lines expected for the run to finish, so it had in the buffer one more line. @roboduck even the non-grbl commands (bCNC intrinsic) I count them, in the lines to be submitted/executed and the gcount. However the WAIT had to be increased after the WAIT was completed.

However I faced another problem, the intrinsic command WAIT waits until the buffer of grbl is empty and grbl reports idle. Funny enough grbl reports for a fraction of a second Idle before the last (1 or 2) probe commands. Maybe @chamnit can enlighten us, how can we know if grbl has emptied all the buffer and it is really idle?

Briefly, now it is fixed (like it was in the past) however it is more safe to have a check on the running state. Sorry for the long wait, it is too cold right now in my garage :)

chamnit, 2016-01-15:
@vladchoudis : This sounds like an unintentional bug in the state reporting. When I get some time, I'll take a look at what happens after a probe cycle. It possible that grbl reports an indeterminant state in this scenario.

roboduck, 2016-01-15:
I may be wrong, but the lines counted in Sender.py:651 don't look like intrinsics (at least in my setup on Win7, python 2.7.2), but like regular grbl commands, but unicode encoded u'string's (therefore isinstance(tosend,str) returns false). Thus those lines are counted 2x, once on line 651 and once when acknowledged by ''ok'' from grbl. Then gcount is probably higher than should and runEnded() is called too early (and then self.running becomes false, therefore following probes are ignored).

However again, I may be wrong, if this is intentional behavior...

roboduck, 2016-01-15:
Sorry, I noticed recent changes, so by Sender.py:651 I meant this line https://github.com/vlachoudis/bCNC/blob/master/Sender.py#L658

What I tried is that I logged gcount anytime it was incremented. In the end of the probing cycle it was more than 30, even when there was only 20-something lines to send (both gcode and bCNC intrinsics).

vlachoudis, 2016-01-16:
@roboduck Maybe if the result of the evaluation L649 returns a unicode it should not be counted. Could you try changing the L654 to
if isinstance(tosend,str) or isinstance(tosend,unicode):
to see if it is ok.

roboduck, 2016-01-16:
@vladchoudis: Yes, I think it make sense, I was thinking about it as well. Going to test it.

roboduck, 2016-01-16:
@vladchoudis: Unfortunately it helps only partly. The reason is that in this case adding empty line ('\n') will also increase number of 'ok' responses from grbl (my is 0.9j.20150930). It will turn into situation when gcount is also incremented (by receiving the 'ok') and the run will end (becase self._gcount >= self._runLines will happen sooner).

Log from run():
```
len(lines) =  14
lines: ['G0Z0.7', 'G0X0Y0', u'G38.3Z-1F1', 'G0Z0.7', 'G0X20Y0', u'G38.3Z-1F1', 'G0Z0.7', 'G0X20Y10', u'G38.3Z-1F1', 'G0Z0.7', 'G0X0Y10', u'G38.3Z-1F1', 'G0Z0.7', 'G0X0Y0']
```
Here is dump from the terminal (cleared just before Autolevel started and removed startup command G90):
```
G0Z0.7
G0X0Y0
G38.3Z-1F1

ok
G0Z0.7
ok
G0X20Y0
G38.3Z-1F1

G0Z0.7
G0X20Y10
G38.3Z-1F1

G0Z0.7
G0X0Y10
G38.3Z-1F1

G0Z0.7
G0X0Y0
[PRB:0.000,0.000,-0.693:1]
ok
ok
ok
ok
[PRB:20.000,0.000,-0.520:1]
ok
ok
ok
ok
[PRB:20.000,10.000,-0.461:1]
ok
ok
ok
ok
[PRB:0.000,10.000,-0.653:1]
ok
ok
ok
ok
```

18x ok
14x G command

in the end (just before runEnded() called _monitorSerial()):
self._gcount: 15
self._runLines: 14

(gcount would probably go up to 18 if not ended)

I hope this will help to analyze to root cause of this issuse.

Btw. absolutely amazing app, thanks! :)

vlachoudis, 2016-01-16:
I am a bit puzzled. Empty lines should be filtered from the "compile" method, unless if they are the result of an expression, which I should fix in the sender. However there is something wrong in your output. The 'ok' are returned by grbl for every command that is send, so it doesn't make sense that the program sends 14 lines and you get 18 'ok'.

roboduck, 2016-01-16:
Maybe I missed something, but if the empty line is added in Sender.py:L655, it goes then direclty to serial, so grbl replies with 'ok'. Then those 4 (=18 - 14) additional 'ok' responses are from the empty lines (or better '\n') which you can see in the output I posted.

vlachoudis, 2016-01-17:
The probe command is taken from the Tkinter widget which returns a unicode instead of a string, causing a mis-sync in the sender. I've added a condition also for the unicode's to be treated as a string.
I've modified also the logic in the autoleveling. Now it counts the probe's that are valid within the given limits and it keeps adding them even after the end of the "run". 

Also I've removed the need to zero after the autolevel cycle. It is up to the user if he considers that he first correctly zeroed the stock before the autolevel. Or to restore the autolevel from a file.
It is needed to avoid tool-change calibration errors, since the location of the zero position is not kept.

roboduck, 2016-01-30:
After trying recent version, I cannot reproduce the issue anymore. Seems that it works OK. Thanks!

vlachoudis, 2016-01-30:
Thank you I close the issue. When grbl will correct the intermittent idle state after the probe I can restore the more secure functionality.

closed 2016-01-30 by vlachoudis