vlachoudis/bCNC #176: Program will not start on virgin PC

flaplante, 2015-12-19:
Attempting to run bcnc.py via Idle on a Windows 7 or Raspberry Pi both yield:
"NameError: name '__file__' is not defined"
change line #46 to read:
try:
    PRGPATH=os.path.sbspath(os.path.dirname(__file__))
except:
    PRGPATH=os.path.sbspath(os.getcwd())

and program starts as expected.

vlachoudis, 2015-12-20:
There should be two underscores before and after \_\_file\_\_.
The \_\_file\_\_ should be the complete filename of the installation location.
The getcwd() returns the executing directory which can be different.

Now why idle is not setting the \_\_file\_\_ I don't understand.
BTW why do you want to run it through idle? and not just executing directly the bCNC.py?

flaplante, 2015-12-21:
Understand. The '_' were there as they should be.  But both the windows 
7 and the Raspberry Pi failed with the same error. Perhaps the issue is 
with Idle. But using getcwd() within the try / except block did the job 
for me.

I use Idle mostly from habit as most of my use of Python is to write new 
code of my own and I am constantly modifying it.  I seldom run into apps 
that others have written in Python. Also I can't seem to get the current 
version of Raspberry Pi to create an icon, so have to use the file 
manager to find and then click on bCNC.py to run without Idle. So for me 
Idle is quicker.

I like bCNC by the way. Faster than GrblController that I had been 
using, and much more features - more than I need actually. I use the CNC 
solely to create small, simple PCB using a Gcode writer that I wrote 
myself (using Python actually).

Fred LaPlante, WA1DLZ

vlachoudis, 2015-12-23:
The getcwd() will only work if you start it from the same directory where bCNC is installed.

flaplante, 2015-12-23:
OK, that is exactly what I did.
Thanks.
Fred LaPlante, WA1DLZ

closed 2016-01-16 by vlachoudis