vlachoudis/bCNC #178: Display does not render 45 degree laser code

atrueresistance, 2015-12-23:
This isn't really a big problem as the code runs fine. bCNC doesn't render gcode exported from picengrave correctly. 

[image]

CamOptics renders
[image]

I used the jtechphonics plugin from inkscape to generate the cutout, and picengrave to do the engrave part. Using the bCNC editor to combine the programs. Must say, love the editor.

[image]

gcode generated
[output_0010.txt](https://github.com/vlachoudis/bCNC/files/71006/output_0010.txt)

Effer, 2015-12-24:
Hi @atrueresistance 

If M3 command is removed (line 31) the render is correct:

[image]

Still to understand the reason  but hope it could help.

Effer, 2015-12-24:
The issue happen every times the M03 is preceding a group of G1/2/3 commands that doesn't explicit the motion modal mode. As example:

G1 X0 Y0
(M03) (Comment/ Uncomment the M3 to see the differences)
X100 Y100 (here we are implicit using G1 )
X50 Y0
G1 X0 Y0

To fix seems enough to comment the line 1028 in CNC.py

```python
1026)    elif c == "M":
1027)       self.mval = int(value)
1028)       self.gcode = None	# Why????
```

But that (# Why????) comment is scaring me a lot. 
So @vlachoudis is the right way?

vlachoudis, 2015-12-28:
You are correct @Effer we should remove this line.

vlachoudis, 2015-12-28:
I've uploaded the correction

closed 2015-12-28 by vlachoudis

Effer, 2015-12-28:
I was almost there.. @vlachoudis  steal my point ;)