vlachoudis/bCNC #182: tool position indicator not accurate to tool path

se7en9057, 2015-12-28:
I have noticed that the arrow indicator for my tool head is not represented accuraitly while running a cut. the tool appears to either be metric or imperial as it moves in the right motions but is far away from the CAM path shown on the screen.

Effer, 2015-12-28:
Hi @se7en9057,
is the cone/arrow behind the green path?
If so, is correct. The green path ahead of the cone represent the buffered work already send to arduino memory.
Also notice that there is a small delay from real machine position and displayed one.

se7en9057, 2015-12-28:
the small cone arrow yes. its far away from any of the green pahts. in all the other software ive used that arrow indicator would be near or on the tool path. but in my case its somewhere else on the screen far away from the tool path. the green progress fill is still moving along the path as expected. its just the tool head indicator thats off on its own somewhere

Effer, 2015-12-28:
That's strange, can you post gcode program and a screenshot?

se7en9057, 2015-12-28:
sure can it will be afew hours but i'll be happy to

vlachoudis, 2015-12-28:
As well your .bCNC configuration file

se7en9057, 2015-12-28:
I'm happy to make anything available to help. The features included with software is some I've been searching for for  some time now. Happy to help

se7en9057, 2015-12-29:
i'm unable to get a screen shot at the moment but i'll update that as soon as i can 

[bCNC]
tool = CNC
page = File

ribbon = File Control Probe Tools Editor Terminal>

file.ribbon = File Pendant Options Close
file.page = DRO Serial

control.ribbon = File Connection User Run Close
control.page = DRO State Control

probe.ribbon = Connection Probe Run
probe.page  = DRO ProbeCommon

tools.ribbon = Database CAM Macros Config
tools.page = Tools*

editor.ribbon = Clipboard Edit Order Move Transform
editor.page = Editor*

terminal.ribbon = Commands Terminal
terminal.page = DRO Terminal*

[Connection]
baud = 115200
port =
pendant = 1
pendantport = 8080
openserial = 0
errorreport = 1

[Control]
step = 1
wcs = 54
#zstep = 1

[Canvas]
view = X-Y
workarea = 1
margin = 1
grid = 1
axes = 1
probe = 1
rapid = 1
paths = 1

[Error]

[Probe]
x =
y =
z =
xmin = 0.0
xmax = 100.0
xn = 5
ymin = 0.0
ymax = 100.0
yn = 5
zmin = -10.0
zmax = 5.0
feed = 10.0
tlo = 0.0
center = 10.0
cmd = G38.2
toolpolicy = 1
toolwait = 1

[File]
dir =
file =
probe =
dxfloopmerge = 0

[Buttons]
n = 7
name.0 = 0
icon.0 =
tooltip.0 = Move to 0, 0, 0
command.0 = G90G0X0Y0Z0
name.1 = scan
icon.1 = margins
tooltip.1 = Move around margins
command.1 = g0 x[xmin] y[ymin]
	g0 x[xmax]
	g0 y[ymax]
	g0 x[xmin]
	g0 y[ymin]

[CNC]
units = 1
lasercutter = 0
acceleration_x = 25
acceleration_y = 25
acceleration_z = 50
feedmax_x = 3000
feedmax_y = 3000
feedmax_z = 2000
travel_x = 200
travel_y = 200
travel_z = 100
round = 4
accuracy = 0.1
startup = G90
spindlemax = 12000
spindlemin = 0
header = M3 S12000
         G4 P3
footer = M5
         G0 Z20

[Font]
dro.status=Helvetica,12,bold
dro.wpos=Helvetica,12,bold
dro.mpos=Helvetica,12

[Color]
canvas.insert  = Blue
canvas.gantry  = Red
canvas.margin  = Magenta
canvas.grid    = Gray
canvas.box     = Cyan
canvas.enable  = Black
canvas.disable = LightGray
canvas.select  = Blue
canvas.select2 = DarkCyan
canvas.process = Green
canvas.move    = DarkCyan
canvas.ruler   = Green
ribbon.active  = LightYellow
ribbon.select  = #c0ffc0

[Shortcut]
F1 = help
F2 = edit
F3 = XY
F4 = ISO1
F5 = ISO2
F6 = ISO3
F7 =
F8 =
F9 =
F10 =
F11 =
F12 =
Shift-F1 =
Shift-F2 =
Shift-F3 =
Shift-F4 =
Shift-F5 =
Shift-F6 =
Shift-F7 =
Shift-F8 =
Shift-F9 =
Shift-F10 =
Shift-F11 =
Shift-F12 =
Control-F1 =
Control-F2 =
Control-F3 =
Control-F4 =
Control-F5 =
Control-F6 =
Control-F7 =
Control-F8 =
Control-F9 =
Control-F10 =
Control-F11 =
Control-F12 =

[Box]
current = 0
n = 1
name.0 = default
dx.0 = 100.0
dy.0 = 70.0
dz.0 = 50.0
nx.0 = 11
ny.0 = 7
nz.0 = 5
profile.0 = 1
overcut.0 = 1
cut.0 = 1

[Stock]
current = 0
n = 1
name.0 = default
material.0 = MDF
safe.0 = 3.0
surface.0 = 0.0
thickness = 5.0

[Cut]
current = 0
n = 1
name.0 = default

[Drill]
current = 0
n = 1
name.0 = default

[Profile]
current = 0
n = 1
name.0 = default

[Pocket]
current = 0
n = 1
name.0 = default

[EndMill]
current = 0
n = 1

_shape.0 = Ball End
_shape.1 = Fishtail
_shape.2 = Radiused
_shape.3 = Square End
_shape.4 = V-cutting

_type.0 = Centercutting
_type.1 = Combination
_type.2 = Down-cutting
_type.3 = Engraving
_type.4 = Non Centercutting
_type.5 = Spiral
_type.6 = Straight
_type.7 = Up-Cutting

_material.0 = Carbide-Tipped
_material.1 = Cobalt
_material.2 = High Speed Steel (HSS)
_material.3 = Powdered Metal (PM) Cobalt
_material.4 = Solid Carbide
_material.5 = Titane

_coating.0 = None
_coating.1 = AlCrTiN
_coating.2 = AlTiCrN
_coating.3 = PCD
_coating.4 = TiAlCrN
_coating.5 = TiAlN Titanium aluminium nitride (black)
_coating.6 = TiCN (bluish)
_coating.7 = TiN Titanium Nitride (yellowish)
_coating.8 = TiN Titanium Nitride (yellowish)

name.0 = Bit1
type.0 = Centercutting
shape.0 = Square End
material.0 = Solid Carbide
coating.0 = None
diameter.0 = 3.175
axis.0 = 3.175
flutes.0 = 4
length.0 = 20
angle.0 =

[Material]
current = 0
n = 13

name.0   = Acrylic
stepz.0  = 0.5
feed.0   = 650.0
feedz.0  = 300.0

name.1   = Aluninum
stepz.1  = 0.1
feed.1   = 127.0
feedz.1  = 60.0

name.2   = Bamboo
stepz.2  = 0.75
feed.2   = 750.0
feedz.2  = 250.0

name.3   = Birch
stepz.3  = 0.75
feed.3   = 750.0
feedz.3  = 250.0

name.4   = Bubinga
stepz.4  = 1.1
feed.4   = 710.0
feedz.4  = 350.0

name.5   = ContraPlaque
stepz.5  = 1.67
feed.5   = 1200.0
feedz.5  = 500.0

name.6   = Copper
stepz.6  = 0.4
feed.6   = 250.0
feedz.6  = 120.0

name.7   = HPDE
stepz.7  = 1.0
feed.7   = 380.0
feedz.7  = 200.0

name.8   = Linoleum
stepz.8  = 1.3
feed.8   = 750.0
feedz.8  = 300.0

name.9   = Maple
stepz.9  = 0.75
feed.9   = 710.0
feedz.9  = 250.0

name.10  = MDF
stepz.10 = 1.67
feed.10  = 1200.0
feedz.10 = 500.0

name.11  = PVC
stepz.11 = 1.3
feed.11  = 750.0
feedz.11 = 300.0

name.12  = Walnut
stepz.12 = 1.0
feed.12  = 810.0
feedz.12 = 400.0

probe.ribbon = Connection Probe Run

Effer, 2015-12-29:
@se7en9057 you've posted the default INI file present in the bCNC folder.
Could you please post the .bCNC file that you can find in your user directory?
Under windows is on  %UserProfile% folder
C:\Users\yourUser\.bCNC

se7en9057, 2015-12-29:
Oh sorry. I'm on Linux btw but I'll try and find it

vlachoudis, 2016-01-30:
@se7en9057 any news on that behavior, before I close the issue?

closed 2016-02-03 by vlachoudis