vlachoudis/bCNC #192: Release Process Considerations

Protoneer, 2016-01-06:
Hi @vlachoudis ,

I am creating a pre-built Raspberry Pi Image files with bCNC included. Could you please consider creating a github release version when you up the version number?

The process has the following benefits:
* It creates a snapshot of the code making it easy to revert to a previous version.
* It automatically creates a zip files containing all of project files, making it easier to package for deployments.
* It will also help with finding regression bugs that might be introduced in new versions. (Isolates new code)

vlachoudis, 2016-01-06:
Good point. I am rather new to git, how can I do it?

Protoneer, 2016-01-06:
Very simple. Go to the release tab, enter the Vxx.xx.xx + release notes and
commit. :)

(Can always create a test repository to first test it with)

carlosgs, 2016-01-20:
It may also be very convenient to have these releases in order to implement the auto-update functionality.
For instance, have a "latest_stable.txt" file in the root of the repository that stores the tag number, then bCNC could compare it, and if necessary fetch the .zip file from the release tab.

vlachoudis, 2016-01-27:
Is there an easy way from a python program to check the latest released tags, instead of looking for a list?

CarlosGS, 2016-01-27:
Here you have a quick-and-dirty program that checks and download the latest version:
```python

import requests
import os
import tarfile

versionCheckURL = "https://github.com/vlachoudis/bCNC/releases/latest"

latestURL = requests.head(versionCheckURL, timeout=5.0 , headers={'Accept-Encoding': 'identity'}).headers.get('location', versionCheckURL)

print("URL for latest version is:" + latestURL)

tag = os.path.basename(latestURL)
print("Latest version is: " + tag)

packageURL = "https://github.com/vlachoudis/bCNC/archive/" + tag + ".tar.gz"

print("Downloading..." + packageURL)
r = requests.get(packageURL)
with open("latest.tgz", "wb") as code:
    code.write(r.content)

print("Extracting...")
tar = tarfile.open("latest.tgz")
tar.extractall()
tar.close()

print("Done!")
```

Hope that helps! :)

CarlosGS, 2016-01-28:
Ok, I didn't even bother looking for the API:
https://developer.github.com/v3/repos/releases/#get-the-latest-release
I'll properly re-implement the example later

CarlosGS, 2016-01-28:
There you go, now using the API:
```python
import requests
import os
import tarfile

__version__ = "0.7.7"
gh_api_url_latest = "https://api.github.com/repos/vlachoudis/bCNC/releases/latest"
update_dir = "./" # should be something like "./updates/"

r = requests.get(gh_api_url_latest)
if r.status_code == 200:
    latest = r.json()
    
    latest_version = latest["tag_name"]
    print("Latest version is: " + latest_version)
    
    if latest_version > __version__ and latest["draft"] == False and latest["prerelease"] == False:
        print("\nThere is a newer version! This will update from " + __version__ + " to " + latest_version + "\n")
        
        packageURL = latest["tarball_url"]
        print("Release name: " + latest["name"])
        print("Description: " + latest["body"] + "\n")
        
        package_file = update_dir + latest_version + ".tgz"
        
        print("Downloading..." + packageURL)
        r = requests.get(packageURL)
        with open(package_file, "wb") as code:
            code.write(r.content)

        print("Extracting...")
        tar = tarfile.open(package_file)
        tar_content = tar.getnames()
        parentFolderName = tar_content[0]
        print("Folder name: " + parentFolderName)
        tar.extractall(update_dir)
        tar.close()

        print("Done!")
else:
    print("There was an error checking for the latest version")
```

vlachoudis, 2016-01-30:
Thank you @CarlosGS. I've just implemented the update feature. For the moment it doesn't download/extract but opens the browser to let the user do it.

closed 2016-01-30 by vlachoudis

CarlosGS, 2016-01-30:
I've seen it, it looks neat, thanks!! :-)