vlachoudis/bCNC #193: After Autolevel, bit travels above workpiece

sjj47, 2016-01-11:
Newbie here.  I've been using bCNC for a while now and I really like it. But I'm running into an odd problem with Autoleveling.   My guess is user-error.

I want to incise a logo into a curved surface - like the edge of a hockey puck. So I set up the object in my vise, covered the surface with aluminum foil, and used the  Autolevel probe feature to digitize the surface. That part works fine. The next step, according to the Github site, is:

*5.Once the probe cycle is finished move the gantry to any location you wish to set the z coordinate as known/set and click the Set Zero. This action will make the z-height map relative to this X/Y location.*

So I move the gantry back to the center of the scanned area, move the bit down until the WCS Z position is 0 -- which has the bit just touching the top of the curve - and click the Probe > Autolevel > Zero button. Then I run the gcode to incise the logo. Invariably the bit travels a few millimeters above the surface I want to cut. 

The bit is obviously matching the surface height, so that part is good, but I can't seem to get the bit down to touch the work. I've tried this several times with the same results. Jogging the (spinning) bit a few mm into the work and then reclicking the Zero button doesn't help. When I execute the gcode the bit moves up and again travels above the surface of the work.

Any suggestions? Thanks,

gcobos, 2016-01-12:
Hi sjj47,

Once auto-leveling is finished, you should use the Probe tool, so the tip will travel once more down until it touches the surface. Then, you have to change the Z by going to "Control" menu and pressing the button with label "Z=0".

As a suggestion, it's good to move the gantry a bit after auto-leveling finishes, so you don't probe into the same point twice (0, 0). There will be a tiny hole already there.

vlachoudis, 2016-01-12:
Just be careful to use the same units everywhere, imperial or metric (what the grbl is reporting in the $nn and what the gcode is using). For the moment the autolevel doesn't accept change of units (imperial/metric) during the sending.

sjj47, 2016-01-13:
Thanks for the responses!   I'll give these suggestions a try and see how it goes.

closed 2016-01-13 by sjj47