vlachoudis/bCNC #194: Italian Translation - Work in Progress

onekk, 2016-01-13:
I don't know how to do it in a better way.

I'm in an advanced stage for an Italian translation, I'm procceding in translate it usign the French Translation as a base (Many thanks to the French translator).

I will not promise an ending date, maybe next week or so (I'm working on it in my spare time).

Regards to all.

Carlo D.

vlachoudis, 2016-01-13:
Thanks @onekk, there is no pressure in time :)

Effer, 2016-01-14:
Hi @onekk 
There's really a huge amount of string to translate, take you time. In the meanwhile I'm working to add some more ;)

I gave a look to the bCNC.po file and a question rise in me. How can you understand where the string is used? Wouldn't be nice to have an option to run bCNC showing the original text and the number of the line to change in the .po file? Maybe we could create a special .po file to use during translation?

onekk, 2016-01-15:
I'm using the french po and modifying it and it containt the source code line references,  with Linux Poedit you can add also comments, and I thinks this is the best way to retain the necessary references.

Having done some other translation works it is however a process of trying and correct what's is wrong, the first attempt is rough but better than nothing, and then refinement over refinement you get a decent work.

Effer, 2016-01-19:
Carlo FIY the other @carlosgs added this tutorial https://github.com/vlachoudis/bCNC/wiki/How%20to%20contribute%20a%20translation
about the translation.

vlachoudis, 2016-01-19:
Thanks @carlosgs

carlosgs, 2016-01-19:
No problem! I'm quite liking the Poedit system, but as @onekk says it would be great to have some context while translating. Not sure if it is possible to do that with the comments, maybe something like _("Cut","text on the button for cut toolpath generation"), _("Cut","text on the button for the clipboard operation") ?
Also, I'm curious to see how flexible Poedit is throughout bCNC's updates (how to merge a previous translation with newly incorporated text).

ThierryM, 2016-01-20:
Hi,
To update your .po file it's quite easy.
First open your translation file .po in Poedit.
Then go to the menu (4th) "Catalog" and choose "Update from a POT File" (sorry is the translation is not precise but I translate from my french GUI)
[image]

Then there is a new window showing the new text chains and the obsolete chains.
[image]

You accept and you can add the new translations. That's all. Regards.

carlosgs, 2016-01-20:
I see, it seems quite convenient. Thanks!!

onekk, 2016-01-23:
Now I'm finish the first "rough" translation starting from the french translation, for reference.

I'm at about line 1800 in the file, I'm editing the file with a text editor and with poedit (search and replace is very helpful for me)

more to come.

@vlachoudis if the po file is some "version" old that cause you some problem?

I'm translating a copy maybe 20 days old.

Regards
Carlo D.

vlachoudis, 2016-01-23:
It won't cause any problem, what ever is not there would appear in English

onekk, 2016-01-23:
!st version is on pull request

Regards

Carlo D.

onekk, 2016-01-23:
Some more work, some errors corrected!

Harvie, 2018-10-29:
Italiano seems to be in place. Closing

closed 2018-10-29 by Harvie