vlachoudis/bCNC #197: multipass plugin

atrueresistance, 2016-01-15:
Hello,

I'm attempting to create a [plugin](https://github.com/atrueresistance/bCNC/blob/multipass/plugins/multipass.py) for bCNC, and if you can't tell I don't normally use python. This is a learning process for me. What is the correct way to append a newblock to the editor? I tried using `app.editor.gcode.blocks`, but there is not `add` method. 

Any helpful comments would be appreciative. I'm not looking for someone to just complete it, as I want to learn more about bCNC and python in general.

vlachoudis, 2016-01-16:
You should use the xxxUndo() methods from the CNC class. They ensure also undo/redo functionality.
To add or append a block on the gcode you can use the "addBlockUndo(pos, block)"
or the "insBlocksUndo(pos, list-of-blocks)"

vlachoudis, 2016-01-17:
You can give also a look on the wiki from @Effer 
https://github.com/vlachoudis/bCNC/wiki/How-to-create-a-plugin

closed 2016-01-17 by vlachoudis

atrueresistance, 2016-01-17:
Thank you for the help, I think I finally got it figured out. You don't want to know how many hours it took, but learned a lot. Attempting to merge plugin in.