vlachoudis/bCNC #198: Loading a file clears auto level data points

asawyer215, 2016-01-16:
I've heard when pcb milling that if you etch before drilling pads you risk listing the pads when drilling. In bCnc if I auto level a section then drill it the auto level data points are cleared but I cannot run the auto level again without risking a leveling point intersecting a drilled hole and failing.

vlachoudis, 2016-01-16:
This is done intentionally for safety. If you want to restore the autolevel data, you would need to save it before loading any other file and reload them afterwards.

asawyer215, 2016-01-16:
@vlachoudis After some experimenting this morning (On the master branch as of 1/16/16) here's what I've found:

1. Save on the autolevel page after zero'ing does not work
2. Save on the autolevel page after the scan is finished opens a save file dialog but the resulting file is 0 bytes long.
3. Hitting save after step two does not work
4. No errors error in the console window log
5. Trying to load the 0 byte file results in an error in the console log (Probably expected)

Am I doing something wrong?

vlachoudis, 2016-01-16:
Strange. I've just tested it and it worked for me both before and after zeroing. Saving is completly independent of zero'ing. If the file is zero bytes long I presume that the autolevel failed somewhere else.

asawyer215, 2016-01-16:
@vlachoudis I just pulled the master branch with the updated auto level changes this morning when I tried it.

carlosgs, 2016-01-16:
@asawyer215 Did you manually typed the .probe extension in the "save" dialog? For instance "surface.probe". The extension is what makes bCNC decide whether to export the gcode/dxf or the probing data

You can also try to export the probing data by using the .stl extension, that should generate a 3D representation of the copper surface. Bear in mind though, you cannot import STL yet, even the ones generated with bCNC's probing capability

asawyer215, 2016-01-16:
@carlosgs 

Hey Carlos!

This is sort of embarrassing - I was trying to save with gcode extension. Switching to .probe got it saving and loading the probe data successfully.

carlosgs, 2016-01-16:
@asawyer215 No worries, I'm glad it is working now :) It is this kind of feedback what helps in order to shape future development and functionalities

I'd appreciate if you can edit the wiki to incorporate what you just learned, to help other people see how to save/import probe files:
https://github.com/vlachoudis/bCNC/wiki/AutoLevel

You can close the issue by pressing the buttons "close issue" or "comment and close" at the bottom of this page

Thanks!! :-)

asawyer215, 2016-01-16:
Thanks Carlos. I added it as a note at the bottom of the wiki.

closed 2016-01-16 by asawyer215

vlachoudis, 2016-01-17:
FYI I have removed the requirement of zero'ing after the autoleveling. It is now the users responsibility that the zero is correct, before or after the autolevel. It is done on purpose in order to keep the calibration data if manual tool change is needed.

Also now if you load a new gcode, it will ask you whether to delete the existing autolevel data or keep them. Which should go to the wiki as well.

massiveradiation, 2020-09-11:
Hello folks, 
Autolevel is amazing and we've been using it with mostly good results, but we're still confused about the process. We're looking for help to clarify the procedure and learn what is happening underneath the hood. 
The procedure on the wiki page is confusing. I think it's important to very carefully distinguish the different between zeroing the Z axis vs hitting the target icon "zero" button in the Autolevel tab, for example.
 
It would be extraordinarily helpful to understand what is happening under the hood.
Here is what we do, with mixed results:
- Load a g-code file .nc
- In the Probe>Autolevel we set X, Y and Z mins and maxs, and probe feed speeds
- We use a normally closed probe device to scan a large flat piece of wood, which is vacuumed down to table. 
- We choose a point on the surface of the wood which is within the area to be scanned and use probe to set Z to zero. At this point, X, Y and Z are all work coordinate zero.
- Then we go to the Probe>Autolevel and hit "scan" button.
- After a successful scan we remove the probe from the router collet and install a bit, then set Z zero at surface of material at X0Y0.
- Then we go to Probe>Autolevel and click the target "Zero" button. We see the height map numbers offset by a small amount at this step. What does pushing this button actually do, adjust all of the height map data points by the new Z0 offset? (if so, why doesn't it do this again when a new g-code file is loaded?)
- Run job and it's usually ok (except with very large files?)
- Now here comes the tricky part, we have multiple operations and want to use the same height map for all of them. So, we load up another g-code file (.nc), and we're asked if we want to delete the autolevel data, we say no.
- We change the tool, re-zero the Z axis
- Now, at this point, do we need to re-click the target "zero" button in the Autolevel area? Is the height map now automatically applied to any g-code file that is subsequently loaded after probing as long as we don't delete it? If we click the target "zero" button again, does it offset the height map twice? Giving us double the correction?
- Also, one of our files is 1.3million lines long, is there an issue running this large of a file with autolevel? It seems to adjust height for the first section of the g-code file, but then stop correcting after a while. 

Thanks so much for hanging in there with this. I'm happy to update the wiki if we can sort it with very concise directions.
Cheers,
Tyler