vlachoudis/bCNC #202: BOX TOOL

Icemanhpg, 2016-01-19:
Hi, firstly great app!! Just starting to get my way around it. Tried to use the box tool after setting up my bits and material  i am a hobbyist with no coding experience , so if my problem seems dumb please excuse my inexperience.. 

The first issue I have is that the fingers of the tool are not of the correct size. I tried the cut, Overcut and profile cuts. Everytime the fingers are too small or too large. In the cut mode the image displayed but still when actual cutting the fingers are way too small and the corresponding part way too big.

I looked at the gcode and what worries me is that it does not seem to take the mill bit size into consideration when making the calculations . i copied the start of my box gcode 

(Block-name: Header)
(Block-expand: 0)
(Block-enable: 1)
M5
(Block-name: Box-Bottom)
(Block-expand: 0)
(Block-enable: 1)
(Box: 100 x 50 x 50)
(Fingers: 5 x 3 x 3)
(  Location: 0,-5.7 )
(  Dimensions: 100,50 )
(  Teeth: 5,-3 )
(  Tool diameter: 0 )
g0 X2.24269 Y-5.7
g1 z0 f250
g1 x22.24269 y-5.7 z0 f700
g1 x22.24269 y0 z0
g1 x42.24269 y0 z0
g1 x42.24269 y-5.7 z0
g1 x62.24269 y-5.7 z0
g1 x62.24269 y0 z0
g1 x82.24269 y0 z0
g1 x82.24269 y-5.7 z0
g1 x102.24269 y-5.7 z0

As you can see it says tool diameter = 0.

is it something I am doing? If so , how do i correct it ( taking in considerations my coding limitation)?

vlachoudis, 2016-01-19:
All the tools are using the information from the databases.
Did you setup the "EndMills", "Materials" database and select the proper "EndMill" and "Stock" material?

Icemanhpg, 2016-01-19:
Yes, the reason for me saying that it is milling the material to the correct depth but the sizes stay of. Even in the graphical interface it shows the fingers are not correct sizes.

I have attached a screenshot of the drawing where you can see the sizes of the fingers.

[image]

In the endmill selection it shows that I have selected my 3.175mm endmill. 

Hope it helps.

vlachoudis, 2016-01-19:
From your image I see that the endmill is taken into account. You can see that edges are "round" with the radius of the endmill, and  there is an overcut depending on the radius of your endmill as well.
Also the dimensions of the box (it is not very clear from the image resolution but still visible) that is your given dimensions + diameter of your endmill.
Finger width depends on your stock thickness setting.

vlachoudis, 2016-01-19:
Just to note, that the Width,Depth and Height if they are positive they refer to the inner dimensions of the box (without including the thickness of the stock). If they are negative they refer to the outer dimensions of the box.

Icemanhpg, 2016-01-19:
My problem is this. With allmy settings confirmed e correct thickness and
endmill selected, no overcut  nor profile cut selected and I tell it to run
the following picture is the result. I did not change any gcode..

Selecting any of the overcut or profile just worsens the issue.

Inam really sorry for inconveniencing you but I am trying to understand
where do I go wrong o is it something else.

PS. I have tried to cut with all 3 different types ( overcut, profile and
just cut) and all 3 results were miserable

Thanks again

Hermann

vlachoudis, 2016-01-19:
I am not sure I understand what you mean. Could you send me your .bCNC (configuration file) as well the settings of the box you are trying to do to understand what is the problem.

Icemanhpg, 2016-01-20:
Hi!

 I have attached my config file and the file I am trying to cut.

I really appreciate the assistance.

[cncboksv3.zip](https://github.com/vlachoudis/bCNC/files/97159/cncboksv3.zip)
[bCNC.zip](https://github.com/vlachoudis/bCNC/files/97157/bCNC.zip)

On another issue Do you have to have GRBL v 0.9j installed for autolevelling?

vlachoudis, 2016-01-20:
Not the system bCNC.ini but the user ini. Typically is found on your home folder as .bCNC
Autolevelling works with version 0.9+. It wont work with 0.8

Icemanhpg, 2016-01-20:
Vasilis

[User .INI.zip](https://github.com/vlachoudis/bCNC/files/97322/User.INI.zip)

Hopefully the right one now.

thanks again for your patience

Hermann

Icemanhpg, 2016-01-20:
Hopefully this explains my predicament better.

In the attached photo you can see that it cuts the size of the fingers
correctly but the size of the endmill in the finger slot creates the
problem that the slot is then basiscally 6mm bigger if I use a 3mm endmill.

Hope this helps

vlachoudis, 2016-01-20:
I've created two boxes one with the profile (and overcuts)
 and one without the profile just to see what you will get.
You have selected an endmill with diameter=3mm and a stock material of thickness=5.8mm

[image]

I've drawn on the plot some dimensions as I measure them with the "Ruler" inside bCNC
As you can see the profile path is offset by diameter/2 = 1.5mm and the fingers are 5.8mm
corresponding the stock material thickness.
The green lines are to show that the fingers are matching perfectly.
Also the interior dimensions of the box are 100x50x40mm

Icemanhpg, 2016-01-20:
Vasilis

Your dimensions are correct. That is what I want. But have a look at the
last photo I sent. That is the crux of my problem. It looks right  ut the
cut differs. I have even gone back to check if my Shapeoko was still
calibrated. It was within 0,1 mm.

I cannot explain the differenc ebetween drawing and the actual cut

vlachoudis, 2016-01-20:
I didn't see any photo?

Icemanhpg, 2016-01-20:
Previous email. Sorry shoud have said so.

vlachoudis, 2016-01-20:
Still I don't understand. What email I didn't receive any email. I see a small email icon next to your post, but clicking it does nothing.

Icemanhpg, 2016-01-20:
Sorry Vasilis. I shows it got sent but ob iously got hung up somewhere.

[image]

As you can see from the photo. It seems as if the fingers are cut correctly but the finger slots are cut 6 mm too big with a 3mm bit.

vlachoudis, 2016-01-20:
Are you sure you are matching the correct sides. The finger size for each side orientation X/Y/Z depends on the number of fingers and the desired dimension. In the image I've send before you can see the green lines I've drawn to show the finger/slot that will be created
and are perfectly matching.

Icemanhpg, 2016-01-21:
That was one of the first things I checked.

Thanks for the patience. I will continue with differentt settings until I
get ut righht

vlachoudis, 2016-01-23:
I am closing this, since it doesn't appear to be an issue of bCNC. However it would be nice to keep us posted if you find the cause of the problem

closed 2016-01-23 by vlachoudis