vlachoudis/bCNC #207: Having troubles making bCNC work on a Mac

murfmv, 2016-01-21:
I have loaded Python 3.5 installed py serial into site packages of 3.5. bCNC still gives me a "Install pyserial error.  BCNC opens fine but cannot accessories the serial. It looks like a very good accomplishment. I need some help!

Effer, 2016-01-21:
Hi @murfmv ,
can you try using python 2.7. There are know problem with python 3.x and unicode strings over serial write. Use latest pySerial version but for python 2.7.

murfmv, 2016-01-21:
Thanks Filippo,

I agonized yesterday getting it to work.  I purposely tried 3.5 because it was newer.  Thanks so much for the answer.  I’m really anxious to try it out!

Fred

Fred Murphy
Lamberts Cove LLC
murfmv@gmail.com
617-335-8610

closed 2016-01-26 by vlachoudis