vlachoudis/bCNC #209: pofile and pocket options

dodolbysuround, 2016-01-22:
I know the issue #139 is closed but don't know where to post...
Maybe I'm missing something but can't find the cut options (surface Z, target depth, depth increment) in the profile and pocket tools... 
May I get a push please?

vlachoudis, 2016-01-23:
For the moment the profile, pocket operation are separate from the cut operation.
So you have first to do a profile or pocket operation
and then select the "Cut" tool and perform a cut operation on the selected paths.

The cut operation takes its parameters if not specified from the "Stock" material

I know that it is a bit counter intuitive or different than other programs, but it was done this way during the development of the program to be easier to debug having clearly separated the steps.

Now I believe I have to include also the cut information in the profile/pocket and in case is left empty to take it from the default... Slowly in the next releases :)

closed 2016-01-24 by vlachoudis

carlosgs, 2016-01-25:
@vlachoudis Great job with the automatic tab functionality, can't wait to mill something with it. It now makes sense to have the "cut" operation separate from the rest :D