vlachoudis/bCNC #212: Messed up settings, program won't start

JaKaspar, 2016-01-23:
Hi, I ran into an interesting problem - i messed up the settings like Acceleration etc. (deleted and left it blank) thinking, that the program will not allow me to leave it like so, and will fix it somehow. But it normally exited, and never started again because of:

line 556, in loadConfig
CNC.acceleration_x = float(config.get(section, "acceleration_x"))
ValueError: could not convert string to 

From the source code (I know nothing about python) I eventually learned, that the configuration is saved in user profile, deleted the files, and problem was gone.

I thing, that the settings should be validated before saving, or when some problem occurs, user should be informed. Also, the corrupted settings could be replaced by those in .ini

Thanks!
Jarda

vlachoudis, 2016-01-24:
Thanks I will put a protection in the code

vlachoudis, 2016-01-24:
Ok now it is protected

closed 2016-01-24 by vlachoudis

JaKaspar, 2016-01-24:
Great, thank you!