vlachoudis/bCNC #213: Applying probe data to multiple gcode files?

sjj47, 2016-01-24:
I've used the bCNC autolevel feature to probe an irregular surface.  The bCNC will apply that probe info to my gcode file when I click the Probe > Autolevel > Zero button.

Is it possible to save the probe information and apply it to multiple gcode files?

I ask because I use F-Engrave to generate vcarve gcode files.  F-engrave normally generates three gcode files:
- main file, milled with a Vbit
- cleanup file, milled with a Vbit
- cleanup file, milled with a straight bit

I'd like to probe a surface, save the probe data (height map?), apply it to the first gcode file,run the job, load the second gcode file, apply the probe data, run the job, load the third gcode file, apply the probe data, and run the third and final job.

Is that possible?   If so, what is the process?

Thanks!

vlachoudis, 2016-01-24:
Yes it is possible. There are two ways. 
1. You can save the probe, from File -> Save using as extension on the file .probe
     Load the new gcode and reload also the probe file
     Change and Zero again the new bit
2. With the current master, if you load a new gcode it will ask you whether to keep or erase the existing probe information if any. Once you change the bit you have to zero again

Option: if you have set up the manual tool change, you don't need to zero again manually.
Click for a manual tool change, change the bit and bcnc will adjust it for you.

closed 2016-01-24 by vlachoudis

sjj47, 2016-01-24:
Thanks for the fast response!   I'll give it a try!