vlachoudis/bCNC #214: current release will not start

flaplante, 2016-01-25:
Just downloaded most recent into Windows 7.
Code of 6 Jan runs fine.
On start get this error:
Traceback (most recent call last):
  File "C:\Program Files\bCNC\bCNC.py", line 2259, in <module>
    application = Application(tk)
  File "C:\Program Files\bCNC\bCNC.py", line 119, in __init__
    self.loadConfig()
  File "C:\Program Files\bCNC\bCNC.py", line 528, in loadConfig
    self.tools.loadConfig()
  File "C:\Program Files\bCNC\ToolsPage.py", line 832, in loadConfig
    tool.load()
  File "C:\Program Files\bCNC\ToolsPage.py", line 317, in load
    self.update()
  File "C:\Program Files\bCNC\ToolsPage.py", line 551, in update
    self.master.cnc()["diameter"] = self.fromMm("diameter")
  File "C:\Program Files\bCNC\ToolsPage.py", line 352, in fromMm
    return self.master.fromMm(float(self[name]))
ValueError: could not convert string to float:

vlachoudis, 2016-01-25:
It seems that you have a non valid float for an endmill diameter. I've put a protection to it now.

flaplante, 2016-01-25:
Thank you sir!

closed 2016-01-25 by vlachoudis