vlachoudis/bCNC #215: USB gamepad as physical pendant

auhopu, 2016-01-26:
Geia sou Vasili,

Here is a low priority feature request, if others find it interesting too:

- add support for a USB gamepad/joystick to perform physical moves

By a quick research in the RPi/RetroPie community, it seems that people use
- joystick, to interface game controllers to Linux
- jtest to identify keystrokes/moves
- pygame to map keystrokes/moves to actions

There's plenty of devices that can be found on eBay for ~10$ (i.e. a lot cheaper than a video editing dial) that could do the trick.

vlachoudis, 2016-01-26:
Geia sou Ilia

hardware pendant is quite interesting to have (there is a discussion that on #155).
There are several approaches:
1. Arduino with firmata
2. Websockets
3. GPIO on Rasberry Pi
4. Xbox controller (look the video on #155)
5. I have a demo connection with a cheap bluetooth serial module and arduino
6. ...

I believe that there is easiest and more flexible is to provide a TCP/IP interface
that once you connect you can send commands to grbl and receive information back.
But without connecting/dropping the connection on every request as it is presently the
web-pendant in bCNC.

So anyone could write a small external program that communicates with his pendant, what ever it is and then it sends the commands to bCNC through TCP/IP.

I can do rather easily a network server for that, but before I was waiting to see if WebSockets could do the same job, which would make also the web pendant more responsive.

auhopu, 2016-01-26:
Sorry, I totally missed the discussion under #155 .

Thinking of it again, your suggestion of providing an open interface via some Web API makes more sense. This way you are not locking your implementation to specific hardware or process.

After reading some examples of pygame like [this] (https://www.piborg.org/joyborg) , the part of capturing the keystrokes is already implemented. The missing glue is how to send them to GRBL, especially when the com port is already in use by an application that communicates with the GRBL board. 

I do not speak German, but it seems that the guy in the Xbox controller video is using xboxdrv and calling a config file. My assumption is that in this file he maps gamepad keystrokes to keyboard shortcuts.

This is an even simpler alternative. Until the new web API is in place, maybe I will try [jkeys](https://code.google.com/archive/p/jkeys/) , another script that is based on [pygame] (https://www.pygame.org/docs/ref/joystick.html) and implements the same joystick-to-keyboard mapping idea.

Btw,  I just read at the GRBL wiki that real jogging will may be implemented in a next release and what is happening now is "pseudo-jogging" with incremental moves.

It is exciting to see ideas coming in on different fronts. Keep it up!

vlachoudis, 2016-01-27:
I am closing this, to keep the discussion in #155

closed 2016-01-27 by vlachoudis